RobTillaart/AS5600 #2: Testing

ahmetcemturan, 2022-05-31:
Hi,
I have uploaded the test.ino file:

It seems to work but not sure about "as5600.readAngle()"

`18:35:42.815 -> 32336	4095	24.53
18:35:43.832 -> 33338	4095	24.53
18:35:44.846 -> 34340	4095	25.05
18:35:45.831 -> 35342	4095	359.38
18:35:46.846 -> 36344	0	96.88
18:35:47.831 -> 37346	0	151.03
18:35:48.848 -> 38348	0	145.41
18:35:49.831 -> 39349	4095	56.00
18:35:50.847 -> 40351	4095	12.48
18:35:51.829 -> 41353	4095	358.15
18:35:52.845 -> 42355	4095	343.91
18:35:53.862 -> 43357	4095	7.12
18:35:54.844 -> 44359	4095	54.95
18:35:55.861 -> 45361	0	94.15
18:35:56.843 -> 46363	0	102.59
18:35:57.860 -> 47364	0	103.56
18:35:58.843 -> 48366	0	103.65
18:35:59.861 -> 49368	0	105.05
18:36:00.844 -> 50370	4095	55.91
18:36:01.863 -> 51372	4095	2.11
18:36:02.880 -> 52374	4095	5.63
18:36:03.863 -> 53376	4095	6.07
18:36:04.880 -> 54377	4095	24.79
18:36:05.863 -> 55379	0	87.65
18:36:06.878 -> 56381	0	101.98
18:36:07.860 -> 57383	0	101.71
18:36:08.877 -> 58385	0	101.89
18:36:09.893 -> 59387	0	98.55
18:36:10.877 -> 60389	4095	54.77
18:36:11.894 -> 61390	4095	15.38
18:36:12.875 -> 62392	4095	15.65
18:36:13.891 -> 63394	4095	64.70
18:36:14.875 -> 64396	0	109.45
18:36:15.892 -> 65398	0	107.43
18:36:16.875 -> 66400	0	105.93
18:36:17.892 -> 67402	0	113.05
18:36:18.907 -> 68403	0	200.53
18:36:19.890 -> 69405	0	202.20
18:36:20.906 -> 70407	0	223.30
18:36:21.886 -> 71409	2722	251.52
18:36:22.896 -> 72411	4095	272.09
18:36:23.912 -> 73413	1763	244.31
18:36:24.895 -> 74414	1148	239.74
18:36:25.910 -> 75416	71	231.74
18:36:26.894 -> 76418	0	228.57
18:36:27.909 -> 77420	0	223.47
18:36:28.926 -> 78422	0	214.24
18:36:29.907 -> 79424	0	205.27
18:36:30.926 -> 80426	0	120.62
18:36:31.908 -> 81428	0	115.08
18:36:32.925 -> 82430	0	100.66
18:36:33.909 -> 83431	0	88.00
18:36:34.925 -> 84433	4095	59.25
18:36:35.942 -> 85435	4095	55.12
18:36:36.922 -> 86437	0	142.15
18:36:37.940 -> 87439	0	196.22
18:36:38.921 -> 88441	0	205.54
18:36:39.939 -> 89443	0	206.68
18:36:40.920 -> 90445	0	206.59
18:36:41.937 -> 91446	0	206.15
18:36:42.953 -> 92448	0	172.92
18:36:43.936 -> 93450	0	126.59
18:36:44.954 -> 94452	0	105.41
18:36:45.935 -> 95454	0	101.54
18:36:46.950 -> 96456	0	82.29
18:36:47.964 -> 97458	4095	57.41
18:36:48.947 -> 98459	4095	34.73
18:36:49.964 -> 99461	4095	8.79
18:36:50.945 -> 100463	4095	350.07
18:36:51.963 -> 101465	4095	336.35
18:36:52.945 -> 102467	4095	336.79
18:36:53.962 -> 103469	4095	334.07
18:36:54.946 -> 104471	4095	334.07
18:36:55.963 -> 105472	4095	334.86
18:36:56.981 -> 106474	4095	332.40
18:36:57.964 -> 107476	4095	332.48
18:36:58.979 -> 108478	4095	30.86
18:36:59.961 -> 109480	0	78.68
18:37:00.979 -> 110482	0	111.56
18:37:01.960 -> 111484	0	137.41
18:37:02.974 -> 112486	0	106.02
18:37:03.987 -> 113487	0	84.31
18:37:04.972 -> 114489	0	85.80
18:37:05.985 -> 115491	0	85.89
18:37:07.001 -> 116493	0	90.46
18:37:07.981 -> 117495	0	90.46
18:37:08.997 -> 118497	0	90.46
18:37:09.980 -> 119499	0	90.46
18:37:11.001 -> 120500	0	90.46
18:37:11.983 -> 121502	0	90.46
18:37:13.000 -> 122504	0	90.46
18:37:14.017 -> 123506	0	90.46

`

ahmetcemturan, 2022-05-31:
And this is the output for the second file:

`18:44:40.724 -> STATUS:	 67
18:44:40.724 -> CONFIG:	 0
18:44:40.724 ->    AGC:	 59
18:44:40.724 -> MAGNET:	 813
18:44:40.724 -> 
18:44:41.707 -> STATUS:	 67
18:44:41.741 -> CONFIG:	 0
18:44:41.741 ->    AGC:	 59
18:44:41.741 -> MAGNET:	 812
18:44:41.741 -> 
18:44:42.722 -> STATUS:	 67
18:44:42.722 -> CONFIG:	 0
18:44:42.722 ->    AGC:	 59
18:44:42.722 -> MAGNET:	 810
18:44:42.722 -> 
18:44:43.738 -> STATUS:	 67
18:44:43.738 -> CONFIG:	 0
18:44:43.738 ->    AGC:	 59
18:44:43.738 -> MAGNET:	 810
18:44:43.738 -> 
18:44:44.720 -> STATUS:	 67
18:44:44.720 -> CONFIG:	 0
18:44:44.720 ->    AGC:	 59
18:44:44.720 -> MAGNET:	 814
18:44:44.720 -> 
18:44:45.737 -> STATUS:	 67
18:44:45.737 -> CONFIG:	 0
18:44:45.737 ->    AGC:	 48
18:44:45.737 -> MAGNET:	 83F
18:44:45.737 -> 
18:44:46.719 -> STATUS:	 67
18:44:46.753 -> CONFIG:	 0
18:44:46.753 ->    AGC:	 42
18:44:46.753 -> MAGNET:	 806
18:44:46.753 -> 
18:44:47.733 -> STATUS:	 67
18:44:47.733 -> CONFIG:	 0
18:44:47.733 ->    AGC:	 44
18:44:47.733 -> MAGNET:	 82A
18:44:47.767 -> 
18:44:48.750 -> STATUS:	 67
18:44:48.750 -> CONFIG:	 0
18:44:48.750 ->    AGC:	 44
18:44:48.750 -> MAGNET:	 82F
18:44:48.750 -> 
18:44:49.732 -> STATUS:	 67
18:44:49.732 -> CONFIG:	 0
18:44:49.732 ->    AGC:	 44
18:44:49.732 -> MAGNET:	 831
18:44:49.732 -> 
18:44:50.749 -> STATUS:	 67
18:44:50.749 -> CONFIG:	 0
18:44:50.749 ->    AGC:	 44
18:44:50.749 -> MAGNET:	 836
18:44:50.749 -> 
18:44:51.732 -> STATUS:	 67
18:44:51.765 -> CONFIG:	 0
18:44:51.765 ->    AGC:	 44
18:44:51.765 -> MAGNET:	 839
18:44:51.765 -> 
18:44:52.747 -> STATUS:	 67
18:44:52.747 -> CONFIG:	 0
18:44:52.747 ->    AGC:	 44
18:44:52.747 -> MAGNET:	 846
18:44:52.747 -> 
18:44:53.764 -> STATUS:	 67
18:44:53.764 -> CONFIG:	 0
18:44:53.764 ->    AGC:	 44
[... 173 lines cut ...]
18:45:28.863 -> CONFIG:	 0
18:45:28.863 ->    AGC:	 4C
18:45:28.863 -> MAGNET:	 828
18:45:28.863 -> 
18:45:29.846 -> STATUS:	 67
18:45:29.846 -> CONFIG:	 0
18:45:29.846 ->    AGC:	 4C
18:45:29.846 -> MAGNET:	 829
18:45:29.846 -> 
18:45:30.863 -> STATUS:	 67
18:45:30.863 -> CONFIG:	 0
18:45:30.863 ->    AGC:	 4C
18:45:30.863 -> MAGNET:	 829
18:45:30.863 -> 
18:45:31.847 -> STATUS:	 67
18:45:31.847 -> CONFIG:	 0
18:45:31.847 ->    AGC:	 4C
18:45:31.880 -> MAGNET:	 829
18:45:31.880 -> 
18:45:32.863 -> STATUS:	 67
18:45:32.863 -> CONFIG:	 0
18:45:32.863 ->    AGC:	 4C
18:45:32.863 -> MAGNET:	 829
18:45:32.863 -> 
`

RobTillaart, 2022-05-31:
> It seems to work but not sure about "as5600.readAngle()"

Think you are right that it needs more testing .

- **readAngle()** seems to give a raw value so there is some BUG.
- **as5600.rawAngle() * AS5600_RAW_TO_DEGREES** seems to give a meaningful angle 

```cpp
void loop()
{
  Serial.print(millis());
  Serial.print("\t");
  Serial.print(as5600.readAngle());
  Serial.print("\t");
  Serial.println(as5600.rawAngle() * AS5600_RAW_TO_DEGREES);
  delay(1000);
}
```

```
comp. time      milis    readAngle    rawAngle*raw2degrees
18:36:58.979 -> 108478   4095         30.86
18:36:59.961 -> 109480   0            78.68
18:37:00.979 -> 110482   0            111.56
```

RobTillaart, 2022-05-31:
> And this is the output for the second file:
Not sure what the values mean..

without much interpretation:

From datasheet, page 18
[image]

```
STATUS: 67   = 0110 0111  =>  MD = 1 = magnet detect     all other bits set are unspecified
AGC:    42   =>  automatic gain control is "nice" between 0x00 and 0xFF
MAGNET: 829 => decimal 2089  on a scale from  0..4095 is halfway the scale
CONFIG: 0 => seems a normal setting. 
```

I will dive into the datasheet somewhere the coming days to see if I missed info for the library. 

Thanks again for testing, I'll let you know if a new version of the library is available.

RobTillaart, 2022-06-26:
@ahmetcemturan 
Created a develop branch (version 0.1.3) with 
- fix in **setConfigure()** 
- option to get angular speed in radians/second

RobTillaart, 2022-07-04:
Released 0.2.0 (almost) with improved **getAngularSpeed()**

I close this issue as there is no recent activity