RobTillaart/AS5600 #4: Feature Request

ahmetcemturan, 2022-06-02:
What would be a great feature is:  Getting the rotation speed of the magnet.

RobTillaart, 2022-06-02:
Good idea:
To determine the rotation speed I expect a circular version of Nyquist should be applied. Normally one has to sample twice the frequency of the sample but for circular movement that is (I assume) not enough as besides the speed one also wants to have the direction.

As the library (on most platforms) cannot run a background thread the only way to determine the rotation speed is to compare the timestamp and position of the last measurement. Note that direction can be indicated by a negative speed.

Something like
```cpp
float getAngularSpeed()
{
  uint32_t now = micros();
  float currentAngle = readAngle();
  
  float angle = currentAngle - lastAngle;
  lastAngle = currentAngle;
  uint32_t deltaT = now - lastMeasurement;
  lastMeasurement = now;

  return (angle * 1e6) / deltaT;  
}
```

RobTillaart, 2022-06-02:
merged into master branch - release 0.1.2

closed 2022-06-02 by RobTillaart