RobTillaart/AS5600 #6: AS5600_COUNTERCLOCK_WISE fails

Shdwdrgn, 2022-06-25:
If you issue `as5600.setDirection(AS5600_COUNTERCLOCK_WISE)` at setup, `as5600.rawAngle() * AS5600_RAW_TO_DEGREES` always returns 337 degrees.  AS5600_CLOCK_WISE works correctly.  I'm just performing a simple measurement for a wind vane so I haven't tried anything more than the basic code.

To work around this for reading CCW, you can find the corrected value with `angle = 360 - angle;` (adjusting for ints, floats, or whatever variable type you use).

RobTillaart, 2022-06-26:
Thanks for reporting this issue and the prposed solution.
The library is experimental as I do not have hardware to test the functionality.
I will investigate / fix later today.

RobTillaart, 2022-06-26:
> If you issue as5600.setDirection(AS5600_COUNTERCLOCK_WISE) at setup, as5600.rawAngle() * AS5600_RAW_TO_DEGREES always returns 337 degrees.

Strange that it returns a constant value, it might be an effect of the unsigned math I use but even then it appears strange.
I will do a careful read through the code of the library to see if I can see a cause. 
(really must buy a sensor someday)

RobTillaart, 2022-06-26:
@Shdwdrgn 

as5600.rawAngle() reads a register from the sensor, no math involved, but it does mask ``` & 0x0FFF```
That is according to the datasheet.

[image]

Does the problem also occur if you use ```as5600.readAngle()```?

RobTillaart, 2022-06-26:
```setDirection()``` should set the **DIR** line of the sensor HIGH or LOW.
Can you measure if this is done correctly ? should equal VDD which can be 5V or 3V3 depending on your setup.
If that voltage is too low it might fluctuate just enough to put the sensor in unknown behavior.
A pullup resistor might be needed.

RobTillaart, 2022-06-26:
@Shdwdrgn 
Found a bug
```cpp
void AS5600::setConfigure(uint16_t value)
{
  writeReg(AS5600_CONF, (value >> 8) & 0x0F);
  writeReg(AS5600_CONF + 1, value & 0xFF);
}
```

should be
```cpp
void AS5600::setConfigure(uint16_t value)
{
  writeReg(AS5600_CONF, (value >> 8) & 0x2F);   //  <<<<<<<<<<< 0x2F
  writeReg(AS5600_CONF + 1, value & 0xFF);
}
```

you might patch that, but as far as I understand the sensor I cannot explain the issue you have with it.

RobTillaart, 2022-06-26:
@Shdwdrgn 
Did review the library code, found an option to optimize and a bug but could not find a cause why CCW creates a constant value of 337 degrees. 

Q: can you post your code that shows this effect?

RobTillaart, 2022-06-26:
Created a develop branch (version 0.1.3) for the bug in **setConfigure()**

Shdwdrgn, 2022-06-26:
I see you were busy while I was sleeping!  For reference, I am using one of the generic square white sensor boards...

Regarding the DIR pin, it appears to be grounded for both options (reads 0.012v on my meter).  I haven't looked and the wiring yet, but it's possible this pin is being pulled low by one of the onboard or internal resistors?

Using the original AS5600_demo sketch and only modifying the setDirection line, I get output such as this:
`65	3839	337.49
1066	3839	337.49
2067	3839	337.49
3068	3839	337.49`

So neither the readAngle value nor the conversion to degrees are changing when I rotate the magnet.  These values do of course change as expected when using AS5600_CLOCK_WISE.

I changed the value to 0x2F in AS5600::setConfigure and restarted the arduino IDE to reload the library, however after compiling the demo again I still get 3839/337.9 values when using AS5600_COUNTERCLOCK_WISE.

Shdwdrgn, 2022-06-26:
OK after a bit of light early-morning reading, I am wondering if the chip you wrote this library for differs from the chip I have here?  
I found some pages from Seeed (https://github.com/Seeed-Studio/Seeed_Arduino_AS5600/blob/master/README.md) which show the same board I have, and they state the DIR line has to be set externally.  I then found the datasheet for the AS5600 (https://ams.com/documents/20143/36005/AS5600_DS000365_5-00.pdf), and page 30 also indicates that direction needs to be physically set with the input pin.  The only reference I can find to direction in the programming section is talking about reading or writing, so it seems that at least with the board I am using, there is no actual option for setting DIR programmatically.

So it would be easy enough to implement this feature in the library through the software conversion, I just don't think it is possible to have the chip itself perform this function unless you apply Vdd to the DIR pin.

RobTillaart, 2022-06-26:
> Regarding the DIR pin, it appears to be grounded for both options (reads 0.012v on my meter). I haven't looked and the wiring yet, but it's possible this pin is being pulled low by one of the onboard or internal resistors?

yes, pulldown resistor to GND.

You should call e.g. **begin(4)** and connect the processor pin 4 to the DIR pin of the sensor. (4 is an example).
Only then the **setDirection()** call will work. (todo for me is to make this clear in the documentation).

----

> OK after a bit of light early-morning reading, I am wondering if the chip you wrote this library for differs from the chip I have here?

The chip is the same, but the breakout box may differ.

----

#### bug found

In the as5600.h file the default parameter of **begin()** is defined as a direction. That is very incorrect.
Will be fixed in 0.1.4 that will be released soon, (I can push it after this msg)

RobTillaart, 2022-06-26:
Note the develop branch is work in progress, so I do not make a PR for it right away.

Shdwdrgn, 2022-06-26:
I just finished setting up the wind sensors on my weather station so I will have to solder up another of the AS5600 boards for testing on my bench... However I just looked at that datasheet again and the actual pin numbers on the chip shows pin 4 and ground, and pin 8 and DIR.  I'll try changing the begin() line in the code to give this a test once I'm set up with another test unit.

Shdwdrgn, 2022-06-26:
as5600.begin(1) always reads clockwise
as5600.begin(2) always reads clockwise
as5600.begin(3) always reads clockwise
as5600.begin(4) always reads 337.49°
as5600.begin(5) always reads 337.49°

Anything above 5 causes a wdt rest on my esp8266.  This is with the current released code, let me load up 'testing' and see if there are any changes.

RobTillaart, 2022-06-26:
You need to connect the DIR pin of the AS5600 chip to an IO Ppin of the microprocessor. 
The IO pin you choose must be given as parameter for the begin call as that is they way the library learns which pin is used to control the DIR pin if the as5600.

Shdwdrgn, 2022-06-26:
Ah that explains a lot!  For my current arrangement I don't have any extra free pins available, I have only hooked up Vcc, Gnd, SCL, and SDA to get the basic I2C functionality.  So is it possible to not pass any value in begin() and avoid the code trying to change one of my already-in-use GPIO pins?

RobTillaart, 2022-06-27:
Mmm, there is no such option yet.
You could try passing -1, that should not interfere with existing io

RobTillaart, 2022-06-27:
Gave it some thoughts and it is possible to handle CW and CCW completely in software / in the library. 
I will make a separate issue for it.

RobTillaart, 2022-06-27:
For the next version (0.2.0 ?) I will investigate software direction control.

That should help solving your problem (took some versions to improve)

RobTillaart, 2022-06-27:
The develop branch is stable now.

Shdwdrgn, 2022-06-28:
Sorry, I haven't had any time to work on this today.  I'll try to get to it tomorrow.

RobTillaart, 2022-06-28:
No problem, time is precious that must be spend wisely

RobTillaart, 2022-07-02:
@Shdwdrgn
Found any time for testing?

Shdwdrgn, 2022-07-02:
Sorry, been a really bad week.  The main sewer line to my house completely plugged up and by the time we got it fixed Thursday I had forgotten all about this.

So back to the subject at hand... When not defining a pin number in the begin() statement this does in fact appear to be working with AS5600_demo_software_direction.ino.  As it switches back and forth between CW and CCW modes I keep rotating the magnet in a CW direction and the output increases or decreases in angle as expected.

I should have some time this evening, so I'll wire up the DIR line to a pin on my esp8266 and verify that the hardware mode also still works.

Additionally, my weather station has been operating all week with a wind vane (the first AS5600 I set up), and that has been operating very reliably, so I don't think there are any major problems with the code for long-term use.  My hardware actually reads in the CCW direction so I'll load up the new code and confirm there are no issues there.

RobTillaart, 2022-07-02:
> The main sewer line to my house completely plugged up and by the time we got it fixed Thursday I had forgotten all about this.

talking about project priorities...

Shdwdrgn, 2022-07-03:
OK I pulled the code from the develop branch that was uploaded 4 hours ago.  The direction control appears to be working via software and with a specified control pin (although I'm not sure if I can tell the difference now since the software method could theoretically be controlling the output under both methods?)  Testing the software method was done with begin().  Testing the GPIO control pin was done with begin(D4) which is the pin I connected the DIR line to.

As far as I can tell, it does appear to be working correctly.  I would give a thumbs-up to merging this code with the master branch.

RobTillaart, 2022-07-04:
I have been thinking of removing the whole hardware pin support as the software direction control just does the job. Reason not to do is that some applications might require signalling e.g led of the cw ccw mode. 

The effect of the sw control on other registers eg range need to be investigated in the future. For now it adds value as is so it is time to release.

Shdwdrgn, 2022-07-04:
I think I would leave it.  As you say, some applications may need external signaling and this provides access to a feature that is already part of the hardware.

One thing that did come to mind is how this and #13 interact with each other.  For example, if you set the direction to CCW, should the offset value be inverted to match the returned angle, or leave it to the user to do the math?  If I set my equipment to point straight forward but the AS5600 returns a value of 10 degrees (when set to CW), then I know I need set an offset of -10 degrees, but that changes if I'm using CCW, and of course it also changes if I'm looking at the top or bottom of the unit so there's a lot of variables that only the designer has control over.  My experience has been that if you try to get too smart in the software, someone will always come along who looks at it differently and still says it's wrong, but I wanted to at least raise the possibility.

RobTillaart, 2022-07-04:
Thanks for raising the point with the CCW. 

from theory it seems it should work.

#### CW 
assume magnet point north (should be 0°)  and as5600 returns 10° (410 on scale 4095)  
Then the user would set offset to -10°

#### CCW
assume magnet point north (should be 0°)  and as5600 returns 350° (3685 on scale 4095)  
Then the user would set offset to +10°

So I expect it to work as is, but the proof is in the pudding test.

Maybe a better way to set the offset is a function **zero()** which is super simple. 
```cpp
void zero()
{
  _offset = 4096 - readAngle();  //  or multiple times and take the average / median
}
```
Call **zero()** if the system is in the orientation that it should be zero and it calculates the offset. 
Assumption is that one can set the system in a zero position and that is not always possible.

The advantage of **setOffset()** is that the user can define which direction / position is zero.
So **zero()** is less functional although easier to understand (assumption).

RobTillaart, 2022-07-04:
@Shdwdrgn
merged develop into master.
If new issues pop up, please create a new issue
Thanks for all your testing, appreciated!

commit 67bc9511 (2022-07-04): Add software direction control (#12)

* fix #6 add software direction control
* fix #14 getAngularSpeed()
* fix #13 add offset functions
* fix conversion constant
* fix readAngle() rawAngle()
* add examples
* update unit test
* update readme.md
closed 2022-07-04 by RobTillaart