RobTillaart/AS5600 #13: Feature request: zero-position offset

Shdwdrgn, 2022-07-02:
In my particular application the AS5600 board does not sit square to the reference point once mounted, so I needed a way in software to compensate for the difference in angle.  I thought this might be a handy feature to include in the library and wanted to share my solution.

`float direction;
float offset = 356.8;

void loop()
{
  int16_t angle = as5600.rawAngle() * AS5600_RAW_TO_DEGREES;
  direction = angle + (360.0 - offset);
  if (direction >= 360) direction -= 360.0;
  if (direction < 0) direction += 360.0;
}`

I assume something like this could be easily implemented in the library as something like setOffset(float offset) to modify the return values of readAngle() and rawAngle(), and there are probably better ways to restrain the values between 0-359.999, but this was a quick solution I put together to get the job done.

RobTillaart, 2022-07-02:
Have made "similar" offset functions in temperature libraries.

Came across a 10 yr old code today, how to make the smallest / fastest look up for windrose directions. 
Like 89° = E and  47° = NE etc

See attached zip for various variations.
[windrose.zip](https://github.com/RobTillaart/AS5600/files/9033472/windrose.zip)

there was an evaluation with it (the mark up is for an old forum, but I (google) cannot find it on the internet anymore)
[numbers.zip](https://github.com/RobTillaart/AS5600/files/9033493/numbers.zip)

RobTillaart, 2022-07-02:
the link was in the code - stupid me :)

https://forum.arduino.cc/t/function-optimization-wind-direction-open-for-ideas/92493/21

RobTillaart, 2022-07-03:
@Shdwdrgn 
The library returns an **unsigned** integer 0..4095 both in readAngle() and rawAngle(). 
This implies that the offset should also be in unsigned format. 

Question: should the offset be in degrees? 
- **void setOffset(float degrees)** converts this to the right offset. 
- **float getOffset()** returns degrees.

I think yes so the user does not need to do the math to convert it to 0 .. +4095

RobTillaart, 2022-07-03:
```cpp
void setOffset(float degrees)
{
  //   floating point modulo handles large values
  if (abs(degrees) >= 360) degrees = fmod(degrees, 360);
  if (degrees < 0) degrees += 360;
  _offset = round(degrees / AS5600_RAW_TO_DEGREES);
}

float getOffset()
{
  return _offset * AS5600_RAW_TO_DEGREES;
}
```

Alternative without the (expensive **fmod()**

```cpp
bool setOffset(float degrees)
{
  if (abs(degrees) >= 360) return false; 
  if (degrees < 0) degrees += 360;
  _offset = round(degrees / AS5600_RAW_TO_DEGREES);
  return true;
}

float getOffset()
{
  return _offset * AS5600_RAW_TO_DEGREES;
}
```

drawback of using integer offset is you loose some precision  (1/8192 max?)

RobTillaart, 2022-07-03:
### Quick test 
test sketch with the fmod() version = 3926 bytes
same sketch without fmod()   = 3240 bytes

delta = 3926 - 3240 = **686** extra bytes for an UNO. (for an UNO that is quite a bit)

Timing
fmod() version 74 us
non fmod() version  4 us

As the offset is set only once (typically) the timing is less an issue,

I am going to check if I can do the % in the integer domain.

RobTillaart, 2022-07-03:
an integer modulo version
```cpp
void setOffset(float degrees)
{
  bool neg = false;
  if (degrees < 0)
  {
    neg = true;
    degrees = -degrees;
  }
  uint16_t offset = round(degrees * (4096 / 360.0)) & 4095;
  if (neg) offset = 4096 - offset;
  _offset = offset;
}
```

```
fmod version   = 4332 (+684)   T = 64 us
bool version   = 3448          T =  4 us
intmod version = 3628 (+180)   T = 32 us
```

This int mod version works up to about ±377.000.000  degrees so should work in almost all practical cases. 

This exercise gave me the insight that the conversion factor should be  360.0 / **4096**;
A full circle is not 4095 so that should be fixed too.

RobTillaart, 2022-07-03:
Conversion constant is fixed in develop branch.

Found another bug in the experimental **getAngularSpeed()** #14 
Needs to be fixed too before 0.2.0 imho.

RobTillaart, 2022-07-03:
Another point.
The offsets will be in degrees only.

RobTillaart, 2022-07-03:
Offset functions are in now. 
the one with integer modulo to save some footprint.

RobTillaart, 2022-07-03:
> This int mod version works up to about ±377.000.000 degrees so should work in almost all practical cases.

update:
Given that an IEEE754 float (degrees) has only 7 significant digits, the practical max is in the order of 100000
Above that number one will have just too few significant digits to do a meaningful conversion / modulo keeping steps of 360/4096 (roughly 1/10th) in mind.

Shdwdrgn, 2022-07-03:
From my own perspective, I'm torn between speed and ease of use.  Perhaps having a 'set' function for both the absolute amount AND for degrees?  But since the back-end code handles everything by the absolute amount (0-4095 I think?) it would appear that the best option is for setDegrees to convert the input to the absolute number, and round it off to the nearest integer.  That creates a small margin of error, but that margin would be constant throughout the entire rotation.  Also I don't see a reason to maintain a float for this offset since I would think the final returned values would still end up being nearly identical.

So my instinct is to give the user the option to set the offset by degrees or absolute value, but only maintain an absolute integer offset to keep the math simple and fast -- which I think is exactly what you did in the last code example above?

RobTillaart, 2022-07-03:
> ... which I think is exactly what you did in the last code example above?

yes, and more important **readAngle()** returns 0..4095 so the (Internal) offset should be able to work with it fast.
Adding the offset is done in only a 1-2 microseconds in one line.

```cpp
uint16_t AS5600::readAngle()
{
  uint16_t value = readReg2(AS5600_ANGLE) & 0x0FFF;
  if (_offset > 0) value = (value + _offset) & 0x0FFF;     // <<<<<<<<<<<<<<<<<
  ...
  return value;
}
```

The develop branch has a lot included now so if you have time and can check if it works as expected ...

I am going to investigate issue #14 as I would like to solve that one too before the merge.

RobTillaart, 2022-07-03:
For completeness I added an example sketch using offset.

Shdwdrgn, 2022-07-04:
This looks great.  I started with your example sketch and the output appeared consistent.  Then I modified the code to simply take a fixed offset value and marked the 'North' position on my magnet for reference.  I needed a -3 degree offset to correctly point to the pin 1 side of the chip, so I tested it with values of -3 and 357, both of which produced identical and correct results as I rotated the magnet (checking positions 0, 90, 180, and 270 roughly by eyeball).  I will recompile my weather station code with the new library and double-check the positioning tomorrow morning when there is no wind, but it looks like this request is resolved.

RobTillaart, 2022-07-04:
Good to hear it behaves stable si far.

closed 2022-07-04 by RobTillaart

RobTillaart, 2022-07-07:
@Shdwdrgn 
there was a serious bug in setConfigure(), fixed in 0.3.0

Shdwdrgn, 2022-07-07:
Thanks, I hadn't seen anything but I'll update and recompile my code to 0.3.0.

commit 67bc9511 (2022-07-04): Add software direction control (#12)

* fix #6 add software direction control
* fix #14 getAngularSpeed()
* fix #13 add offset functions
* fix conversion constant
* fix readAngle() rawAngle()
* add examples
* update unit test
* update readme.md