RobTillaart/AS5600 #14: getAngularSpeed() fails when over 360 degrees

RobTillaart, 2022-07-03:
```cpp
  int      angle   = readAngle();
  uint32_t deltaT  = now - _lastMeasurement;
  int      deltaA  = angle - _lastAngle;   //  fails mod 360
```
 if lastAngle == 355 and angle == 2  the deltaA should be 7 not -353

Need a construction that takes into account that we do not know if the thing runs CW or CCW.

To be investigated a.s.a.p  so it can be part of 0.2.0

Shdwdrgn, 2022-07-04:
Testing this with the updated code in the develop branch from 5 hours ago, I think this one is resolved.  I spun the magnet back and forth across the 359/0 degree threshold multiple times and the values from getAngularSpeed() remained generally consistent within the limitations of my manual movements.

RobTillaart, 2022-07-04:
Thanks for testing this one too

commit 126a77c1 (2022-07-03): fix #14 getAngularSpeed()
commit 4fdca1fc (2022-07-03): fix #14 getAngularSpeed()
commit 67bc9511 (2022-07-04): Add software direction control (#12)

* fix #6 add software direction control
* fix #14 getAngularSpeed()
* fix #13 add offset functions
* fix conversion constant
* fix readAngle() rawAngle()
* add examples
* update unit test
* update readme.md
closed 2022-07-04 by RobTillaart