RobTillaart/AS5600 #19: Support for other I2C addresses/AS5600L

Stathis1991, 2022-08-11:
The AS5600L is a newer variant of the AS5600, it supports burning a new I2C address, and it also has a different default I2C address (0x40 instead of 0x36). Would it be possible to add a feature that allows you to pass a I2C address other than 0x36, so that multiple sensors can be used without multiplexing?

Stathis1991, 2022-08-11:
If interested, I could also send you a custom test PCB that contains a working AS5600L with a QWIIC connector.

RobTillaart, 2022-08-11:
Thanks for the issue, 
Do you have a link to the datasheet ?
Yes interested in such PCB. Please contact me by email (see json file)

Stathis1991, 2022-08-11:
Definitely, the datasheet of the AS5600L can be found here:
https://ams.com/documents/20143/36005/AS5600L_DS000545_3-00.pdf/7ade6878-7a32-2294-b88d-479d50fab6de

The chip is very similar to the AS5600 (it works perfectly with this library when the default address is changed to 0x40), the major difference is that this version is oriented towards I2C usage with support for multiple sensors on the same bus.

I will send you a separate email for the AS5600L PCB!

RobTillaart, 2022-08-11:
@Stathis1991 
Created a AS5600L branch - https://github.com/RobTillaart/AS5600/tree/AS5600L 
It is a starting point for the support of the AS5600L specific functions.
For now the version number = 0.3.1, might become 0.4.0

TODO
- real life testing
- unit tests
- readme.md update
- example sketch
- whatever more is needed.

RobTillaart, 2022-08-12:
@Stathis1991

#### Progress

Updated the AS5600L branch - https://github.com/RobTillaart/AS5600/tree/AS5600L

Please verify if it works for you.

How the address can be changed temporary and have 2 or more AS5600L's on one bus is not clear from datasheet.
Switching them on in a specific order seems the only way (not nice).

#### Differences

Major differences found so far:
- support for I2C address setting
- removed support for ANALOG OUT !
- part of permanent data can only be written 2x instead of 3x

#### TODO
- real life testing
- example sketch I2C address change

Stathis1991, 2022-08-12:
Really happy with this, thanks! I will test it soon.

Regarding the I2C address, from what I understand in the datasheet it is only possible to burn a new permanent I2C address for the chip, temporary changes are not possible. And indeed these I2C address burns can be executed only twice.

Stathis1991, 2022-08-12:
It appears that I was wrong, both temporary and permanent address changes are possible. According to the datasheet (p28):

Temporary Change: Write new slave address into register
I2CADDR and I2CUPDT. This change of the I²C slave
address is just temporary. After power-on reset the slave
address is set to previous value.
• Permanent Change: Write the new slave address to
register I2CADDR, then perform BURN_SETTING
command. This option is for permanently changing the I²C
slave address.

RobTillaart, 2022-08-13:
> Regarding the I2C addresses, it indeed appears to be inconvenient to configure temporary addresses, and would potentially require extra pins to configure the chips while they are still on 0x40.

In some projects it is not allowed to switch on all actuators/sensors at once as that could peak current / drop voltage.
I imagine that with an PCF8574 (I2C 8 bit IO) the VCC of a number of sensors is controlled. Switch them on in a defined order and give them their new address before switching on the next sensor. Expect you can configure 10++ sensors per second. 

**I2CUPDT**
For me it is not clear what I2CUPDT is. Best idea so far is that it stands do I2C_UPDATE. What do you think?

RobTillaart, 2022-08-27:
@Stathis1991 
Little time to work on this one, will maybe be next week.

RobTillaart, 2022-08-30:
@Stathis1991 
Found time to solder the connector today and started testing.
- fixed the **setAddress()** for the AS5600L. It needed the address also in the I2CUPDT register.
- added an AS5600L specific test sketch 
- updated the documentation

All updated code can be found in branch here - https://github.com/RobTillaart/AS5600/tree/AS5600L

I will test some more today and if I find bugs or tests fail I will add fixes and let you know.

RobTillaart, 2022-08-31:
@Stathis1991 
All fixes are in as far as my tests and reading went. So I will merge later today / tomorrow.

closed 2022-09-01 by RobTillaart