RobTillaart/AS5600 #21: detectMagnet always false

DP1000, 2022-10-04:
Hi everyone, i've tried the AS5600_demo example, added a simple
`if(as5600.detectMagnet()) Serial.print("TRUE\t");
  else Serial.print("FALSE\t");`
In the loop, but it always gives me false.

RobTillaart, 2022-10-04:
Thanks for the issue,
You should know that the magnet should be quite close to the sensor.

Note: magnetic fields fade with cubed(distance ) - much faster than e.g gravity which fades with squared(distance)

DP1000, 2022-10-05:
Hi @RobTillaart , i've just tried and you're right,
yesterday it was giving me false because i had the magnet in an enclosure that kept it away approx. 1cm.
Sorry for bothering.

closed 2022-10-05 by DP1000

RobTillaart, 2022-10-05:
No need to apologize, this is the way people learn.