RobTillaart/AS5600 #32: What is the Maximum Rotation Speed ​​in RPM?

rtek1000, 2023-02-26:
Hello,

I was able to successfully test the position sketch, but offset deviation is occurring with each course reversal.

I'm trying to read the motor shaft position from a motorized potentiometer. Ref.: https://a.aliexpress.com/_mrwDFDs

When manually rotating the axis slowly, the position offset does not deviate. But if it's too fast, a detour occurs.

Being the 12-bit AS5600, and the 150us sampling. How many rpm is the rev limit?

If each lap has 4096 positions. Does each position need at least 150us to be read?

RobTillaart, 2023-02-26:
Thanks for the issue,
I am in the middle of some bug fixes (other lib) so it can take some time to dive into this one.

The AS5600 is in the first place a position(orientation) sensing device as it is reasonable fast it can be used for RPM too.
Q1: What is the max RPM you want to measure?

For now I can only say that I did no extensive performance measurements, so I have no 100% clear answer.

You can win some performance by increasing the I2C bus speed. It should support up to 1000 kHz.
Q2: what processor do you use?

RobTillaart, 2023-02-26:
@rtek1000 

Did a small I2C measurement (ESP32 @240 MHz) based upon AS5600_I2C_frequency.ino (added time measurement).

```
CLOCK   us      raw   degrees    //  added for clarity  Clock = i2c clock
100000	640	748	65.74
200000	404	788	69.26
300000	325	787	69.17
400000	286	788	69.26
500000	262	788	69.26
600000	245	788	69.26
700000	235	788	69.26
800000	226	788	69.26
900000	226	788	69.26
1000000	226	788	69.26
```

I2C clock from 100k => 400k you win factor ~2.2
I2C clock from 100k => 800k you win factor ~2.8

(simplified estimate @ 400 kHz)
Assume you use 15% of the time to sample and 85% for processing the sample etc
Then you have a sample every 2.5 milliseconds ==> 400 samples per second.
You need ~4 samples per rotation => max 100 rotations per second 
So 6000 RPM should be feasible (within assumptions)

https://github.com/RobTillaart/AS5600/issues/24 confirmed measurements up to 1400 RPM.

rtek1000, 2023-02-26:
I'm using ESP8266 at 160MHz, and 400kHz clock on I2C. I tried 800kHz but my AS5600 crashed for some unknown reason.

I believe the potentiometer is doing about 1800 RPM. Cursor displacement needs 5.9 motor shaft revolutions in 200ms. About 30 rotations per second.

RobTillaart, 2023-02-26:
> ESP8266 at 160MHz and 400 kHz clock on I2C.

So processor / I2C is fast enough.  

6 shaft revolutions in 200ms == 1 revolution in 30 ms

so you need to sample every 7 ms or so.

Can you test how fast the ESP8266 can sample the sensor?
```cpp
//
//    FILE: AS5600_I2C_frequency.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//    DATE: 2022-05-28

#include "AS5600.h"
#include "Wire.h"

AS5600L as5600;   //  use default Wire

uint32_t clk = 0;

uint32_t start, stop;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  Wire.begin();

  //  UNO
  as5600.begin(4);  //  set direction pin.
  //  ESP32
  as5600.begin(14, 15);     // no direction pin.
  as5600.setAddress(0x40);  // AS5600L only

  as5600.setDirection(AS5600_CLOCK_WISE);  // default, just be explicit.
  int b = as5600.isConnected();
  Serial.print("Connect: ");
  Serial.println(b);
}

void loop()
{
  clk += 100000;
  if (clk > 500000) clk = 100000;
  Wire.setClock(clk);
  
  delay(10);
  start = micros();
  int angle = as5600.readAngle();
  stop = micros();
  
  Serial.print(clk);
  Serial.print("\t");
  Serial.print(stop - start);
  Serial.print("\t");
  Serial.print(x);
  Serial.print("\t");
  Serial.println(as5600.rawAngle() * AS5600_RAW_TO_DEGREES);

  delay(1000);
}

// -- END OF FILE --
```

rtek1000, 2023-02-26:
> > ESP8266 at 160MHz and 400 kHz clock on I2C.
> 
> So processor / I2C is fast enough.
> 
> 6 shaft revolutions in 200ms == 1 revolution in 30 ms
> 
> so you need to sample every 7 ms or so.
> 
> Can you test how fast the ESP8266 can sample the sensor?
> 
> (you need to adapt the pins)
> 
> ```c++
> //
> //    FILE: AS5600_I2C_frequency.ino
> //  AUTHOR: Rob Tillaart
> // PURPOSE: demo
> //    DATE: 2022-05-28
> 
> 
> #include "AS5600.h"
> #include "Wire.h"
> 
> AS5600 as5600;   //  use default Wire    
> 
> uint32_t clk = 0;
> 
> uint32_t start, stop;
> 
> void setup()
> {
>   Serial.begin(115200);
>   Serial.println(__FILE__);
>   Serial.print("AS5600_LIB_VERSION: ");
>   Serial.println(AS5600_LIB_VERSION);
> 
>   Wire.begin();
> 
>   //  UNO
>   as5600.begin(4);  //  set direction pin.
>   //  ESP32
>   as5600.begin(14, 15);     // no direction pin.
>   as5600.setAddress(0x40);  // AS5600L only
> 
>   as5600.setDirection(AS5600_CLOCK_WISE);  // default, just be explicit.
>   int b = as5600.isConnected();
>   Serial.print("Connect: ");
>   Serial.println(b);
> }
> 
> 
> void loop()
> {
>   clk += 100000;
>   if (clk > 500000) clk = 100000;
>   Wire.setClock(clk);
>   
>   delay(10);
>   start = micros();
>   int angle = as5600.readAngle();
>   stop = micros();
>   
>   Serial.print(clk);
>   Serial.print("\t");
>   Serial.print(stop - start);
>   Serial.print("\t");
>   Serial.print(x);
>   Serial.print("\t");
>   Serial.println(as5600.rawAngle() * AS5600_RAW_TO_DEGREES);
> 
>   delay(1000);
> }
> 
> // -- END OF FILE --
> ```

The IDE was unable to compile this sketch:

Line 30:
`as5600.setAddress(0x40);  // AS5600L only`

> 'class AS5600' has no member named 'setAddress'; did you mean 'getAddress'?

rtek1000, 2023-02-26:
I did another test, I put the maximum speed that my circuit can give to the engine (L293D IC), about 5900 RPM (60ms). Strangely the displacement graph seems to be much more stable.

The main difference is the use of PWM for motor speed control. In the displacement of 60ms, the voltage of 9V is applied directly to the motor. In the displacement of 200ms, the motor receives energy via PWM. I believe that the AS5600 is receiving electromagnetic interference from the motor.

[image]
[image]

RobTillaart, 2023-02-26:
>  class AS5600' has no member named 'setAddress'; did you mean 'getAddress'?

I have an AS5600L - that is a derived class, you can remove that line.

RobTillaart, 2023-02-26:
> I believe that the AS5600 is receiving electromagnetic interference from the motor.

mmmm think think think

You use a 8266, can you add a call to **analogWriteFreq(new_frequency)** in setup().

The default frequency is 1 kHz (?)  - can you set it to 10.000 Hz. ==> **analogWriteFreq(10000)**

max is 40 kHz - https://randomnerdtutorials.com/esp8266-pwm-arduino-ide/

rtek1000, 2023-02-26:
> mmmm think think think
>
> You use a 8266, can you add a call to **analogWriteFreq(new_frequency)** in setup().
>
> The default frequency is 1 kHz (?) - can you set it to 10.000 Hz. ==> [...]

Actually, I'm using an Arduino Nano (ATmega328) to control the motor, and an ESP8266 to read the AS5600. The PWM is sourced from pins D5 and D6, it should be around 960Hz. Ref.: [analogwrite](https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/)

RobTillaart, 2023-02-26:
Can you use a 490Hz PWM pin of the nano?  
just to see if there is a difference?

2) Are the nano and ESP connected, e.g. do they share GND or other signals?

RobTillaart, 2023-02-26:
Need to attend other tasks - to be continued

rtek1000, 2023-02-26:
The original idea was to adapt the AS5600 to a Yamaha mixer (01v96), but at the moment I don't have this table at hand. The potentiometer track wears out with use, so I thought of adapting the AS5600 together with an MCP4725 to replace the analogue potentiometer track. The problem is that the Yamaha mixer makes rapid movements of the potentiometer cursors, so the response time needs to be small, otherwise the mixer must oscillate the potentiometer cursor. I believe the mixer uses PID to control the motor. In another mixer model (Pro Mix 01), where the potentiometer cursor was very easy to move (very worn), the mixer was oscillating the potentiometer cursor as well.

rtek1000, 2023-02-26:
> Can you use a 490Hz PWM pin of the nano? just to see if there is a difference?
> 
>     2. Are the nano and ESP connected, e.g. do they share GND or other signals?

I will test with 490Hz, yes the grounding is common.

rtek1000, 2023-02-26:
Using pins D9 and D10 (490Hz) the offset deviation also occurred, in the same way as with pins D5 and D6 (980Hz).

I'll have to wait to have access to the mixer to test the original motor control circuit, to find out if the AS5600 will be affected. I remember that eventually the 01v96 potentiometer made a noise in the order of kHz, so it must be similar.

Anyway, thanks for the feedback.

About another sketch, of angular velocity, I noticed that the RPM is smaller than the W.

Is the W in radians? Shouldn't it be the other way around, W less than RPM?

rtek1000, 2023-02-26:
> About another sketch, of angular velocity, I noticed that the RPM is smaller than the W.
> 
> Is the W in radians? Shouldn't it be the other way around, W less than RPM?

In the source code it has a reference of 180 degrees, does this mean that there is no way to use this function for full turns (>360 degrees)?

rtek1000, 2023-02-26:
I did some more tests, added an LC filter to the motor, but it didn't fix it.

Then I removed the timing belt from the engine pulley and noticed that the graph was stable.

Apparently, when reaching the end of the potentiometer travel, the timing belt is jumping from the engine pulley, (which is not good).

In this way it must not be an electrical problem, but a mechanical one.

I'm going to try to add a sensor at the end of the stroke to shut off the motor before the timing belt jumps off the pulley.

RobTillaart, 2023-02-27:
> Apparently, when reaching the end of the potentiometer travel, the timing belt is jumping from the engine pulley, (which is not good).
In this way it must not be an electrical problem, but a mechanical one.

Think you've almost found the root cause. Question remaining is why the belt is jumping. 
- too large?
- not tight enough?
- wear and tear?
- motor too fast?
- does it heat up (and expand just a tiny bit)
- motor need readjustment?

RobTillaart, 2023-02-27:
> About another sketch, of angular velocity, I noticed that the RPM is smaller than the W.
Is the W in radians? Shouldn't it be the other way around, W less than RPM?

which sketch exactly? There might be an error in the math. 

From theory:
1 RPM == 360°/minute == 6.28.. radians/ minute ==> RPM < radians/minute < degrees /minute

And if angular velocity is defined in seconds than  
1 RPM == 6°/second == 0.104.. radian/sec  ==> radians/s < RPM < degrees/s

RobTillaart, 2023-02-27:
looked at the code and the raw measurement is converted with these constants:
```cpp
//  0.087890625;
const float   AS5600_RAW_TO_DEGREES     = 360.0 / 4096;
//  0.00153398078788564122971808758949;
const float   AS5600_RAW_TO_RADIANS     = PI * 2.0 / 4096;
//  4.06901041666666e-6
const float   AS5600_RAW_TO_RPM         = 60.0 / 4096;
```

1 rotation per second == 4096 raw ==>
- 360 degrees per second
- 2 PI radians per second
- 60 RPM 

So if the unit RPM is chosen you get a unit per **minute**, otherwise you get per **second**.

rtek1000, 2023-02-27:
> Think you've almost found the root cause. Question remaining is why the belt is jumping.
>
>     * too large?
>
>     * not tight enough?
>
>     * wear and tear?
>
>     * motor too fast?
>
>     * does it heat [...]

I made two virtual sensors for the end of strokes, which gave good results.

The ESP sends a signal to the NANO when the position is within the range of 1500 for the end of the stroke, in this way the motor is turned off before the cursor hits the housing too hard.

Apparently the problem is that the timing belt and the pulley are already very used, but the idea is to take advantage of the potentiometers in this state.

For you to have an idea, the new potentiometer track is 10k, but when the cursor is failing, the track resistance is already at 20k many times. Aging causes an increase in the ohmic resistance of the track.

This ohmic resistance drift causes a shift in the relative position of the mixer cursor, the cursor stops on a different scale position.

I believe the next step is to add MCP4725, to try to make NANO obey random position commands, like stopping the cursor at intermediate positions.

About the maximum RPM reading of the AS5600, I still don't know what it is, but it must be higher than the value I got (>5900 RPM).

rtek1000, 2023-02-27:
> which sketch exactly? There might be an error in the math.
>
> From theory: 1 RPM == 360°/minute == 6.28.. radians/ minute ==> RPM < radians/minute < degrees /minute
>
> And if angular velocity is defined [...]

Using an online calculator, 1800 RPM is about 188.5 rad/sec.

[Calculator Angular speed to RPM](https://calculator.academy/rpm-to-angular-velocity-calculator/) 

I used the Angular speed examples.

> getAngularSpeed()
> getAngularSpeed(AS5600_MODE_RPM)

RobTillaart, 2023-02-27:
> Using an online calculator, 1800 RPM is about 188.5 rad/sec.

yup factor = 2PI / 60 ~0.1047...    (or reverse 60/2PI = 9.55..)

There are many ways to express angles - wrote this converter library some months ago as a spin of 
- https://github.com/RobTillaart/AngleConvertor
(Note that rotation is mentioned as "turns", maybe add two oneliners to add "rotations" for . )

RobTillaart, 2023-02-27:
> About the maximum RPM reading of the AS5600, 
I still don't know what it is, but it must be higher than the value I got (>5900 RPM).

There is no hard number, it depends on accuracy and precision one wants.
If we assume that during a measurement of 200 usec the magnet may only move 1 raw position  (ultra high accuracy).
At that speed the sensor would only move ~1.2 rotations per second = 72 RPM 

However
If we assume that during the measurement of 200 usec the magnet may move 100 or even 1000 raw positions. 
Then the RPM goes up with the same factor. However the accuracy will drop 

#### From what we know:

From measurements above (400000	**286**	788	69.26)  we see that can make a measurement in 286 usec over I2C.
For 1 rotation one needs at least 2+ measurements per rotation. With 2 measurements you loose the information about direction. So lets say we make 5 measurements in 2 rotations (that is 2.5 measurement per rotation, enough to determine the direction of the rotation).
In one second we can make 1.000.000 / 286 ~ 3500 measurements. 
5 needed to make 2 rotations => 3500 / 5 * 2 = 700 * 2 = 1400 rotations per second.
1400 * 60 = 84000 RPM 

84000 RPM, think you will get mechanical problems too (heat, wear/tear)

More realistic you want to do more than 100% of the time making measurements. 
So in practice I expect 10000 - 15000 RPM is pretty much the limit.

For faster RPM measurements one uses e.g. white and black patterns on the axis. 
Shine a light on it and a photosensor generate pulses which are counted in an interrupt routine. 
Depending on the pattern one get 1 .. n pulses per rotation. 
Assume 1 pulse per rotation and an interrupt routine using 1 us (ESP) one could measure up to several million RPM.

rtek1000, 2023-02-27:
> There is no hard number, it depends on accuracy and precision one wants. If we assume that during a measurement of 200 usec the magnet may only move 1 raw position (ultra high accuracy). At that [...]

The operating range of the AS5600 really seems to be very wide.

Just out of curiosity I found a 500,000 RPM [motor](https://www.celeroton.com/en/products/motors/).

I don't even know where to use a 500,000 RPM engine. For drone use maybe?

RobTillaart, 2023-02-27:
Dentist drill?

rtek1000, 2023-02-28:
Thank you for your attention!

closed 2023-02-28 by rtek1000

RobTillaart, 2023-02-28:
You're welcome