RobTillaart/AS5600 #33: How to use Wire1?

kingjamez, 2023-03-07:
I'd like to use 2 AS5600's on seperate I2C interfaces. All of the examples call AS5600 with the "default Wire".
I tried: AS5600 as5600(&Wire1) with no luck.

RobTillaart, 2023-03-07:
Thanks for the question
I will look into it asap but might take a few days.

RobTillaart, 2023-03-07:
One option to consider is is to use the AS5600L which can use two addresses.

RobTillaart, 2023-03-07:
What board do you use?
Which begin call do you use?
The begin(sda,scl) call sets a hard Wire

RobTillaart, 2023-03-07:
I have written a minimal sketch which compiles for the **NANO33_BLE** and the **TEENSY_4.1**but not tested with hardware.
Both have a second I2C named Wire1.

```cpp
//
//    FILE: AS5600_demo_two_I2C.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo two I2C busses
//    DATE: 2023-03-07
//
//  Works only if Wire1 bus is present
//  - nano33 ble
//  - teensy 4.1

#include "AS5600.h"
#include "Wire.h"

AS5600 as5600_0(&Wire);
AS5600 as5600_1(&Wire1);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  as5600_0.begin(4);  //  set direction pin.
  as5600_0.setDirection(AS5600_CLOCK_WISE);
  Serial.print("Connect device 0: ");
  Serial.println(as5600_0.isConnected() ? "true" : "false");
  delay(1000);
  
  as5600_1.begin(5);  //  set direction pin.
  as5600_1.setDirection(AS5600_COUNTERCLOCK_WISE);
  Serial.print("Connect device 1: ");
  Serial.println(as5600_1.isConnected() ? "true" : "false");
  delay(1000);
}

void loop()
{
  Serial.print(millis());
  Serial.print("\t");
  Serial.print(as5600_0.readAngle());
  Serial.print("\t");
  Serial.print(as5600_1.readAngle());
  Serial.print("\n");
  delay(100);
}

//  -- END OF FILE --
```

kingjamez, 2023-03-07:
Thank you very much for the very quick help! After seeing your code I knew that my original try of AS5600 as5600(&Wire1) should have worked. Indeed the problem was that I was using a Rapberry Pi Pico and while it is present, Wire1 is not defined in the default pins_arduino.h. After adding the additional lines for SDA1 and SCL1 for Wire1, I then used your code unchanged and was able to talk to the AS5600 boards on the two built in I2C channels. So everything works great! Thanks again for the very fast and responsive help. I'm using these AS5600 boards with your library and the RPi Pico board in a DIY flightsimulator and it's working extremely well.

RobTillaart, 2023-03-07:
> Thank you very much for the very quick help! 

You're welcome.

Good to hear you found the solution, maybe you should report the missing I2C port with your solution to the pico people.

> I'm using these AS5600 boards with your library and the RPi Pico board in a DIY flightsimulator and it's working extremely well.

Good to hear, photos are welcome!

kingjamez, 2023-03-07:
I'll absolutely post some photos once it's done. Working on finishing the handle and it's buttons now, but that is trivial compared to getting excellent resolution on the limited travel X and Y axis. 

I reported my findings on the Arudino Pico github. Hope it helps someone in my position: https://github.com/arduino/ArduinoCore-mbed/issues/194#issuecomment-1458532012

-Jim

RobTillaart, 2023-03-07:
@kingjamez 
1) think you should open a separate new issue for it to put it explicit into the spotlight.
2) when posting a piece of cpp you should use the syntax highlighting trick to look like a pro (see your post, open it with edit)

kingjamez, 2023-03-07:
Thanks! I thought I did the code syntax, but mistakenly just clicked the code button not knowing how it worked, it just added two apostrophes and I put my code between them... obviously that was wrong. :-) Got it figured out now, thanks for spurring me to get that right!

RobTillaart, 2023-03-08:
As the issue is solved I close it

closed 2023-03-08 by RobTillaart