RobTillaart/AS5600 #34: Weird noise in cumulative position reading

cranefist, 2023-05-07:
I have this weird noise on the cumulative position output:

[image]

[image]

The output is :

readAngle() / rawAngle()
getCumulativePosition()

Also, simply reading and printing out readAngle() considerably lessens the frequency of the noise.

Also, it seems you are using rawAngle output for the cumulativeposition, wouldn't it be better to use the filtered readAngle output? So i could choose the filter settings? Now my filter settings have no effect on the cumulative output.

RobTillaart, 2023-05-07:
Thanks for this issue,
Think I understand the point you make.
Quite busy so will try to look into this coming week.

cranefist, 2023-05-07:
Ok, ill keep investigating this. As your library now uses the exact same code i was using for the cumulative position. Ill try to see if i get the same noise using my old code.

Really nice that you added it btw. Simplified things a lot.

The noise is still there, its really weird. I thought i solved it, but nope. And that noise cant come from the unfiltered raw output. As i dont see the same noise on that output.

RobTillaart, 2023-05-07:
> wouldn't it be better to use the filtered readAngle output? 
So i could choose the filter settings? 
Now my filter settings have no effect on the cumulative output.

Think it would be better, need to read the data sheet again if there are reasons pro/con.

Note: Would also need to change **resetCumulativePosition()**
It would bring code in line with **getAngularSpeed()** which uses **readAngle()** too.

cranefist, 2023-05-07:
Ok, good that you mentioned that resetCumulative also needed to be changed, i now switched to the filtered output and will see how it goes.

I did have some issues of values bouncing and the turn point. That can really reek havoc. I remember when i was at this same problem with you at some point, when i was trying to figure out why i was missing points at the turn around point. And its exactly the hysteresis, and its there for a purpose.

I use these both for motors, and dials. So for dials i want to use max slow filters so they are as smooth as possible. And for motors faster filters for fastest possible output.

RobTillaart, 2023-05-11:
Merged and released 0.3.7.
So you should now be able to use the filters, 
If there are still problems feel free to reopen this issue (or open a new topic)

closed 2023-05-11 by RobTillaart