RobTillaart/AS5600 #36: AS5600 module came with a resitor between PGO and GND

Zipdox2, 2023-05-16:
I spent several hours trying to change the output mode to PWM. Both with the library and bare I2C. Actually, the output wasn't working at all. I discovered that the module has a resistor between PGO and GND, and that it was causing the AS5600 to disable the output (to use it for programming, check datasheet). This resistor needs to be removed to use the out pin.

This obviously has nothing to do with the library, but I think it could save some people a lot of trouble if you would add a disclaimer/warning somewhere in the README (under the OUT pin section) for people to check whether this is the case, and to remove the resistor if they intend to use the out pin.

RobTillaart, 2023-05-16:
Thanks for this info, I will add it ot the readme

RobTillaart, 2023-05-16:
Added section below in readme.md file.
Will be part of the next version.

Thanks again for this info. 

----

##### Note: (From Zipdox2 - See issue #36)
Some AS5600 modules seem to have a resistor between **PGO** and **GND**.
This causes the AS5600 to disable the output (to use it for programming, see datasheet). 
This resistor needs to be removed to use the **OUT** pin.

closed 2023-05-16 by RobTillaart