RobTillaart/AS5600 #37: Connecting multiple encoders

MaksymKotwas, 2023-05-26:
I'm working on my robotics project and I'm using a library to check the position of stepper motors. Can you help me and tell me how to connect two or more encoders to a single arduino board?

RobTillaart, 2023-05-26:
Hi, thanks for your question.

The datasheet states that the I2C address of the AS5600 is always **0x36** so that gives a small problem.
If you need only two you can replace one by an AS5600L which has address **0x40**.
These two could be on one I2C bus, without problem. 

If you want to have more there are two options
- use of multiple I2C busses (e.g. software I2C)
- use an I2C multiplexer.

The first option (SW I2C) can give problems as it is hard to make it reliable and fast.

The second option, the use of a I2C multiplexer is easier and you could use the https://github.com/RobTillaart/TCA9548.
This gives you 8 channels, so you can use 8 AS5600's with the same address and select the one you want to use,
of even 16 is you put one AS5600 and one AS5600L behind each channel.

As you can have 8 **TCA9548** on the I2C bus (address 0x70..0x77) the maximum is 64 AS5600 or even 128 devices if you also add AS5600L's on each channel. (you need an additional power supply for all the devices)

The drawback is that (1) you need to select the right channel every time you want to read a sensor and (2) it is slightly slower due to the extra I2C commands. 

If performance allows the I2C multiplexer is the way to go.

If performance must be high, you might consider two - https://github.com/RobTillaart/HC4067 - in parallel for the SDA and SCL lines. Gives you selection from 16 (32) devices.

MaksymKotwas, 2023-06-01:
Hi I have one more question, 
j3.begin(3);
this line is for encoder to be connected to analog pin 3 what should I do if I have my multiplexer connected to scl and sda instead?

RobTillaart, 2023-06-01:
Not worked with analog out myself, so this is not verified or tested.

Every AS5600 should have its own analog input. So either you give all of them an unique analog pin or add a multiplexer between the analog port too. 
Note that you only need to set all devices in analog mode and use tha analog port multiplexer.

Check my HC40xx multiplexer libs. 

How many as5600 do you need?

MaksymKotwas, 2023-06-01:
for now I need 5 encoders and I bought dfrobot i2c multiplexer

RobTillaart, 2023-06-01:
Think one HC4067 should be enough to multiplex the analog inputs..

MaksymKotwas, 2023-06-01:
I have enough analog inputs but I don't know how I should connect as5600 to i2c multiplexer and to analog input at the same time

RobTillaart, 2023-06-12:
> I don't know how I should connect as5600 to i2c multiplexer

https://www.bluedot.space/tutorials/connect-multiple-sensors-using-i2c-multiplexer/

> and to analog input at the same time

Use the 6 analog ports of an Arduino.

If this does not help, better ask your questions on the Arduino forum for project guidance. 

----

As this problem is not related to this library related, I close the issue.

closed 2023-06-12 by RobTillaart