RobTillaart/AS5600 #38: Burn functions not working?

cedricjeanty, 2023-05-31:
I have been working with your library and a handful of AS5600s to try and program them to generate a PWM output signal with permanently burned zero and max angle settings. I can get the PWM to work, and can confirm the settings I write are working, but when I try to burn them (CONF first, and then ZPOS, MPOS). The angles seem to change after burning. I had to uncomment the burn functions, and correct a small typo. Do you have experience burning the angles? What did you do to make it work?

RobTillaart, 2023-05-31:
Thanks for opening this issue.
I have never burned the angles so no experience.
Unfortunately this means I don't know the answer

Can you share the typo, so I can fix?

cedricjeanty, 2023-05-31:
The register write values were "x0x40". They had an extra hex prefix.
Thanks for your help

RobTillaart, 2023-05-31:
> The register write values were "x0x40". They had an extra hex prefix.

OK, 
I added that one (in fact two) as an extra safety so people must change it after uncommenting.
This reduces the chance of people using it by "accident".
(so it will stay).

----

I had a quick look in the datasheet and saw there are two methods to burn the angle.
- I2C
- OUT pin

Which method did you use?

----

Datasheet (page 21)

_Burn_Angle Command (ZPOS, MPOS)
The host microcontroller can perform a permanent
programming of ZPOS and MPOS with a BURN_ANGLE
command. To perform a BURN_ANGLE command, write the
value 0x80 into register 0xFF. The BURN_ANGLE command can
be executed up to 3 times. ZMCO shows how many times ZPOS
and MPOS have been permanently written.
**This command may only be executed if the presence of the
magnet is detected (MD = 1)**._

Did you check if the magnet was see?

----

Datasheet (page 22)

_Step 7. Verify the BURN_ANGLE command:
Write the commands 0x01, 0x11 and 0x10 sequentially into the register 0xFF to load the actual
OTP content._

The library has not support for this step directly as **writeReg()** is private,
By making this function public you should be able to do this step.

Although verify does not sound like a mandatory step, it might be

cedricjeanty, 2023-05-31:
I saw that as well. Maybe I'll try to add that to the burn function, as
well as a verification.

cedricjeanty, 2023-06-01:
Well, I finally figured it out. It turns out that it is important to write the MANG register when you burn the config since they are burned together in the burn conf command. Then, you can burn the ZPOS. In my case I was trying to burn the CONF with MANG set to zero, and then set both ZPOS, MPOS and then burn angles. That does not appear to work. The correct order is:

set configs
set mang
burn conf+mang
reboot
set zpos
burn angles

I hope this helps someone.

RobTillaart, 2023-06-01:
Some questions pop up
- did you use the verify steps?
- do you have a minimized sketch that can be used?
- after burning the angle, can one still (temporary) overrule the ZPOS?

cedricjeanty, 2023-06-01:
Good questions. 
I added the verify steps, but I have not tested if they are necessary:
void AS5600::burnSetting()
{
  writeReg(AS5600_BURN, 0x40);
  delay(5);
  writeReg(AS5600_BURN, 0x01);
  writeReg(AS5600_BURN, 0x11);
  writeReg(AS5600_BURN, 0x10);
  delay(5);
}

My sketch is a monster right now. I have not minimized it yet. I need to move on to the next step in my project...

You can still re-write temporarily, and then burn new ZPOS (3 times) after the config burn. The important step I had not initially realized is that if you want to burn the config, it appears you must write a MANG, and it will be permanently written.

RobTillaart, 2023-06-01:
Thanks for these additions, very valuable.
When time permits I will update the readme with this information..
Think I can write a minimized version based upon your info.
To be continued (next week I hope to have some time)

cedricjeanty, 2023-06-01:
I'd be happy to review and test it.

RobTillaart, 2023-06-18:
@cedricjeanty 
Took some time but wrote two sketches:
- one for writing settings and max angle
- one for writing zpos

Some lines need to be uncommented for safety. See comments in sketches.

There is at least one improvement to make.
- readback the settings to check correct values are written before burning

I created a branch https://github.com/RobTillaart/AS5600/tree/burntest 
Will upload the sketches there too.

Please let me know your comments.

```cpp
//
//    FILE: AS5600_burn_conf_mang.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//    DATE: 2023-06-18

//  WARNING
//  As burning the settings can only be done once this sketch has to be used with care.
//
//  You need to 
//  - uncomment burnSettings() in AS5600.h and AS5600.cpp.
//  - adjust settings and MaxAngle in burn_mang() function below ==> line 77++
//  - uncomment line 107

#include "AS5600.h"

#include "Wire.h"

AS5600 as5600;   //  use default Wire
//  AS5600L as5600;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  //  ESP32
  //  as5600.begin(14, 15);
  //  AVR
  as5600.begin(4);  //  set direction pin.
  as5600.setDirection(AS5600_CLOCK_WISE);  // default, just be explicit.

  if (as5600.isConnected())
  {
    Serial.println("Connected");
  }
  else
  {
    Serial.println("Failed to connect. Check wires and reboot.");
    while (1);
  }

  Serial.println("\nWARNING  WARNING  WARNING  WARNING  WARNING  WARNING\n");
  Serial.println("This sketch will burn settings to your AS5600.");
  Serial.println("Adjust the settings in the sketch to your needs.");
  Serial.println("Press any key to continue.");
  Serial.println("\nWARNING  WARNING  WARNING  WARNING  WARNING  WARNING\n\n");
  while (Serial.available()) Serial.read();
  while (!Serial.available());
  Serial.read();

  while (Serial.available()) Serial.read();
  Serial.print("Are you sure to burn settings + maxangle? [Y for Yes]");
  while (!Serial.available());
  char c = Serial.read();
  if (c == 'Y')
  {
    burn_mang();
  }

  Serial.println("\nDone..");
}

void loop()
{
}

void burn_mang()
{
  bool OK = true;
  OK = OK && as5600.setPowerMode(0);
  OK = OK && as5600.setHysteresis(0);
  OK = OK && as5600.setOutputMode(0);
  OK = OK && as5600.setPWMFrequency(0);
  OK = OK && as5600.setSlowFilter(0);
  OK = OK && as5600.setFastFilter(0);
  OK = OK && as5600.setWatchDog(0);
  OK = OK && as5600.setMaxAngle(0);
  if (OK == false)
  {
    Serial.println("\nERROR: in settings, burn_mang() cancelled.");
    return;
  }

  Serial.println();
  Serial.println("burning in 5 seconds");
  delay(1000);
  Serial.println("burning in 4 seconds");
  delay(1000);
  Serial.println("burning in 3 seconds");
  delay(1000);
  Serial.println("burning in 2 seconds");
  delay(1000);
  Serial.println("burning in 1 seconds");
  delay(1000);
  Serial.print("burning ...");
  //  uncomment next line
  //  as5600.burnSettings();
  Serial.println(" done.");
  Serial.println("Reboot AS5600 to use the new settings.");
}

// -- END OF FILE --

```

and 

```cpp
//
//    FILE: AS5600_burn_conf_mang.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//    DATE: 2023-06-18

//  WARNING
//  As burning the settings can only be done once this sketch has to be used with care.
//
//  You need to 
//  - uncomment burnSettings() in AS5600.h and AS5600.cpp.
//  - adjust settings and MaxAngle in burn_zpos() function below ==> line 77++
//  - uncomment line 100

#include "AS5600.h"

#include "Wire.h"

AS5600 as5600;   //  use default Wire
//  AS5600L as5600;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  //  ESP32
  //  as5600.begin(14, 15);
  //  AVR
  as5600.begin(4);  //  set direction pin.
  as5600.setDirection(AS5600_CLOCK_WISE);  // default, just be explicit.

  if (as5600.isConnected())
  {
    Serial.println("Connected");
  }
  else
  {
    Serial.println("Failed to connect. Check wires and reboot.");
    //while (1);
  }

  Serial.println("\nWARNING  WARNING  WARNING  WARNING  WARNING  WARNING\n");
  Serial.println("This sketch will burn settings to your AS5600.");
  Serial.println("Adjust the settings in the sketch to your needs.");
  Serial.println("Press any key to continue.");
  Serial.println("\nWARNING  WARNING  WARNING  WARNING  WARNING  WARNING\n\n");
  while (Serial.available()) Serial.read();
  while (!Serial.available());
  Serial.read();

  while (Serial.available()) Serial.read();
  Serial.print("Are you sure to burn zpos? [Y for Yes]");
  while (!Serial.available());
  char c = Serial.read();
  if (c == 'Y')
  {
    burn_zpos();
  }

  Serial.println("\nDone..");
}

void loop()
{
}

void burn_zpos()
{
  //  ADJUST ZPOS
  if (as5600.setZPosition(0) == false)
  {
    Serial.println("\nERROR: in settings, burn_zpos() cancelled.");
    return;
  }

  Serial.println();
  Serial.println("burning in 5 seconds");
  delay(1000);
  Serial.println("burning in 4 seconds");
  delay(1000);
  Serial.println("burning in 3 seconds");
  delay(1000);
  Serial.println("burning in 2 seconds");
  delay(1000);
  Serial.println("burning in 1 seconds");
  delay(1000);
  Serial.print("burning ...");
  delay(1000);
  //  uncomment next line
  //  as5600.burnAngle();
  Serial.println(" done.");
  Serial.println("Reboot AS5600 to use the new settings.");
}

// -- END OF FILE --
```

RobTillaart, 2023-06-18:
@cedricjeanty 
Updated sketches in burntest branch.

RobTillaart, 2023-06-27:
@cedricjeanty 
Going to merge burntest branch, 
will mention code as experimental.

RobTillaart, 2023-06-28:
@cedricjeanty 
I close this issue as the code has been merged (as experimental).
Feel free to reopen the issue if needed, or create a new one.

Thanks for the insights in the "burn process".

closed 2023-06-28 by RobTillaart

M4RL0NN, 2023-08-14:
Hello, I'm assembling the AS5600 sensor programming code on my arduino UNO.

when I request the compilation of the error code on this line:

// as5600.burnSettings();
"Compilation error: 'class AS5600' has no member named 'burnSettings'

// as5600.burnAngle();
"Compilation error: 'class AS5600' has no member named 'burnAngle'; did you mean 'rawAngle'?"

...

M4RL0NN, 2023-08-14:
Hello, I'm assembling the AS5600 sensor programming code on my arduino.

when I request the compilation of the error code on this line:

// as5600.burnSettings();
"Compilation error: 'class AS5600' has no member named 'burnSettings'

// as5600.burnAngle();
"Compilation error: 'class AS5600' has no member named 'burnAngle'; did you mean 'rawAngle'?"

RobTillaart, 2023-08-14:
@M4RL0NN 

To get access to the burn commands you need to uncomment them.
This is done as a safeguard as the burning are permanent and cannot be undone (afaik)
Please read the datasheet (twice) so you understand the implications.
Using this library for burning the sensor is at your own risk (although you can see above in this thread that it should work).

AS5600.h file you need to uncomment some lines so it looks like:
```cpp
  //  BURN COMMANDS
  //  DO NOT UNCOMMENT - USE AT OWN RISK - READ DATASHEET
  void burnAngle();
  void burnSetting();
```

AS5600.cpp file you need to uncomment:
```cpp
/////////////////////////////////////////////////////////
//
//  BURN COMMANDS
//
//  DO NOT UNCOMMENT - USE AT OWN RISK - READ DATASHEET
//
void AS5600::burnAngle()
{
   writeReg(AS5600_BURN, x0x80);
}

//  See https://github.com/RobTillaart/AS5600/issues/38
void AS5600::burnSetting()
{
  writeReg(AS5600_BURN, 0x40);
  delay(5);
  writeReg(AS5600_BURN, 0x01);
  writeReg(AS5600_BURN, 0x11);
  writeReg(AS5600_BURN, 0x10);
  delay(5);
}
```

JabberwockPL, 2023-09-13:
I have burned the zero and max angles, no verification was required, I just used:
  as5600.setZPosition(z);
  as5600.setMPosition(m);
  as5600.burnAngle();

Abi-64, 2023-09-22:
I successfully permanently changed the address on AS5600L using this code:
  ASL.setAddress(100);
  delay(100);
  ASL.burnSetting();

Thanks for the library!!

RobTillaart, 2023-09-22:
Thanks for reporting, appreciated!
Good to hear it works!

Abi-64, 2023-09-22:
Thanks @RobTillaart! btw, in this line of code there is one extra x before 0x80. 
 writeReg(AS5600_BURN, x0x80);

RobTillaart, 2023-09-22:
> Thanks @RobTillaart! btw, in this line of code there is one extra x before 0x80. writeReg(AS5600_BURN, x0x80);

It is an extra safety net to prevent accidental burning :)

Abi-64, 2023-09-22:
oh I see! :)
I just changed the I2C address for another IC, this time I didn't use any specific circuit mentioned in the datasheet, I just connected 4 pins(v3.3,GND, SDA and SCL directly to the Arduino Uno pins! this means one can change the IC address using a  Socket Adapter SOIC8 SOP8 like this one:
https://www.amazon.ca/Socket-Adpter-Programming-Adapter-Module/dp/B0892F713P/ref=sr_1_1_sspa?keywords=soic8+clip&qid=1695410710&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

I hope it would be helpful.

Chandu07-dot, 2023-10-18:
C:\Users\SS\AppData\Local\Temp\arduino\sketches\CD03BF04600DA408FABC0FBF8D3F40B3\sketch\AS5600_burn_zpos.ino.cpp.o:(.literal._Z9burn_zposv+0x2c): undefined reference to `AS5600::burnAngle()'
c:/users/SS/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\SS\AppData\Local\Temp\arduino\sketches\CD03BF04600DA408FABC0FBF8D3F40B3\sketch\AS5600_burn_zpos.ino.cpp.o: in function `burn_zpos()':
C:\DW\AS5600-master\AS5600-master\examples\AS5600_burn_zpos/AS5600_burn_zpos.ino:105: undefined reference to `AS5600::burnAngle()'
collect2.exe: error: ld returned 1 exit status

I am getting the above error while uploading in Arduino ide, I made the changes mentioned in previous comments.
provide me solution

I just want to know first I have to execute this AS5600_burn_zpos.ino file and then AS5600_burn_conf_mang.ino
to make zero position to burn and I want to do max position to burn how I do it .

Thank you in advance.

RobTillaart, 2023-10-18:
>  C:\DW\AS5600-master\AS5600-master\examples\AS5600_burn_zpos/AS5600_burn_zpos.ino:105: undefined reference to `AS5600::burnAngle()'
collect2.exe: error: ld returned 1 exit status

The IDE caches libraries.
So it could be that you uncommented the functions while the IDE was open.
Please try to close all instances of the IDE and try again (just try build without upload first)

> I just want to know first I have to execute this AS5600_burn_zpos.ino file and then AS5600_burn_conf_mang.ino
to make zero position to burn and I want to do max position to burn how I do it .

you best check the datasheet on this. From my head I thought it is order independent, however check the datasheet.

RobTillaart, 2025-07-07:
@eriknz 
Good to hear you have it working. 👍 

As the zpos et al can only be written once or a few times, I decided to make it not possible by accident. This can be annoying I understand, however solvable when reading the documentation as you did. 

Frankly I do not know why the sensor "calibration" can only be written once.