RobTillaart/AS5600 #39: Bug with using second I2C bus on ESP32 for two AS5600 encoders

jtran03, 2023-06-27:
I have been trying to implement two AS5600's on the same ESP32 using its two I2C bus interfaces. However, after running this code here: 
```
#include "AS5600.h"
#include "Wire.h"

const int SCL_1 = 23; 
const int SDA_1 = 22;
const int SCL_2 = 19;
const int SDA_2 = 18; 

AS5600 as5600_0(&Wire);
AS5600 as5600_1(&Wire1);

void setup()
{
  // Initialise Serial Connection
  Serial.begin(115200); 

  // Initialise as5600_0 Connection
  as5600_0.begin(SDA_1, SCL_1); 
  Serial.print("Connect device 0: ");
  Serial.println(as5600_0.isConnected() ? "true" : "false");
  
  // Initialise as5600_1 Connection
  as5600_1.begin(SDA_2, SCL_2); 
  Serial.print("Connect device 1: ");
  Serial.println(as5600_1.isConnected() ? "true" : "false");
}

void loop()
{
  Serial.print(millis());
  Serial.print("\t");
  Serial.println(as5600_0.rawAngle() * AS5600_RAW_TO_DEGREES);
  Serial.print("\t");
  Serial.println(as5600_1.rawAngle() * AS5600_RAW_TO_DEGREES);
  Serial.print("\n");
  delay(100);
}
```

It caused the 2nd AS5600 is to copy the readings of the 1st AS5600. I found that temporary fix is to comment out line 68 of AS5600.cpp: 

`// _wire = &Wire;`

RobTillaart, 2023-06-27:
I will look into this issue asap, but it might take a few days.

RobTillaart, 2023-06-27:
Will merge -https://github.com/RobTillaart/AS5600/pull/40 first today.

RobTillaart, 2023-06-27:
@jtran03 
As I have no hardware setup nearby, are you able to do some testing when I provide a new branch?

RobTillaart, 2023-06-27:
> Will merge -#40 first today.

done

RobTillaart, 2023-06-27:
>  I found that temporary fix is to comment out line 68 of AS5600.cpp:

Thanks
Your analysis is correct, it has been in the code since initial release.
New release will be 0.4.0 as this is serious bugfix.

jtran03, 2023-06-27:
> @jtran03 As I have no hardware setup nearby, are you able to do some testing when I provide a new branch?

Yep

RobTillaart, 2023-06-27:
Created develop branch and a PR - https://github.com/RobTillaart/AS5600/pull/41

- minimal code change, only removed that one line.
- includes minor edit in readme.md  about pull ups when using an UNO.

I'll wait for your test results, thanks!

jtran03, 2023-06-27:
Hi, 

Tested both sensors using as5600.readStatus() and as5600_0.readAngle and it seems to be working fine

RobTillaart, 2023-06-28:
Thanks, will merge and release

commit 319985dc (2023-06-28): fix #39 (#41)

- fix #39 support for Wire2 on ESP32
- update readme.md
closed 2023-06-28 by RobTillaart