RobTillaart/AS5600 #43: Oscillations in speed measurements

JabberwockPL, 2023-07-13:
I have looked through other issues and while some are similar, none are exactly the same.

I am trying to measure two things on my stationary bike: speed and total distance. I have made an initial setup and while the readouts for angles and rotations seem to work fine, speed gives regular oscillations, like here:

[image]

and here:

[image]

The difference between the two is in the delay value: 10 ms and no delay.

RobTillaart, 2023-07-13:
Thanks for reporting this issue. Quite busy so it might take a few days before I can dive into it.

RobTillaart, 2023-07-13:
Can you post a minimal sketch that shows this effect?

Please be aware that the AS5600 is primary a position sensor and the library "uses the delta-position as speed".

JabberwockPL, 2023-07-13:
The graph is from the example sketch on angular speed, the only change I have made was to the delay value. The example on angular speed RPM gives the same shape of the curve with different values.

RobTillaart, 2023-07-13:
OK, thanks
Can you add timestamps in micros() to the sketch and provide a CSV of 20 ms?

JabberwockPL, 2023-07-13:
OK, I will do that tonight. However, I have found one possible cause: as I wrote my setup is somewhat temporary and when I have adjusted the position of the sensor, the curve definitely changed. That makes sense: if the magnet is not perfectly aligned to the center of rotation and the sensor to the magnet, the positions and rotations would be more or less correct, but the angular speed on different sections would vary... The problem is differences are really minimal, so it would have to be aligned to fractions of a milimeter.

RobTillaart, 2023-07-13:
mmm, interesting, 
please continue to add your insights / observations. (I'll try to come back later this evening)

JabberwockPL, 2023-07-13:
For the time being I had to take down my support, I need to come up with someting more stable (and possibly adjustable).

RobTillaart, 2023-07-14:
> The difference between the two is in the delay value: 10 ms and no delay.

(thinking out loud)

The 10 ms delay picture shows  2 sinus waves / frequencies
- slow = 2 waves in 500 samples
- fast = ~42 waves in 500 samples
==> ratio is 1 to 21

The no delay picture shows 
- slow ~11 waves
- fast ~ first slow wave has also around 21 waves, remainder look similar
==>  similar ratio

So independent of sampling frequency there seems to be a "harmonic 21x as fast"
Is there some gear or involved in your setup.

Is it possible to run the setup on battery only?
I often disconnect my laptop from wall socket to get rid of 50 Hz creeping in over USB power

JabberwockPL, 2023-07-14:
The actual setup is as messy electrically as it can be, as it has, beside the ESP32 module, a game racing wheel connected to a computer, and a stationary bike braked electromagnetically, supplied from the mains, so to be honest, I only half expect it to work... 

_However_, being aware of that, for testing I was just using the ESP32 module with my laptop with all the rest disconnected. I cannot recall whether the laptop was powered from the mains or from the battery, I will check it out when I fix the mounting support. But of course, it is still possible something in the vicinity is still giving the interference. Unfortunately, the bike is _really_ stationary, moving it around is quite problematic.

Before I have taken down the support, I have tested it with another standalone ESP32 (Wemos LOLIN32 Lite) as I run out of Arduinos (it was the second Pro Micro that I have broken the connection off... :( ) and the results were similar, although I cannot be sure the curve had the exact same parameters.

The 'fast' waves sometimes appeared even when the magnet was stationary, but they often corrected themselves, like the chip was self-adjusting. However, for my purposes I can ignore them - if I convert the speed to the racing wheel accelerator/brake, it will be negligible. The 'slow' waves, unfortunately, are quite visible. I will see whether better placement of the magnet/sensor will correct that...

JabberwockPL, 2023-07-14:
I have made a better support (but it still might be about a half of milimeter off in one direction, I am working on that) AND I have disconnected all nearby devices and run the test from the battery: the 'slow' wave seems to be gone. The 'fast' one remains, as can be seen here:
[report.csv](https://github.com/RobTillaart/AS5600/files/12050998/report.csv)

I cannot figure out how the 'fast' wave relates to the rotation of the magnet...

And sorry for the wall of text above, I supose the reports are to be terse...

RobTillaart, 2023-07-14:
Sorry , glitch of my hand on the phone closed the issue for a second.

RobTillaart, 2023-07-14:
### report.csv

Had a look at the samples 6000-6100 and noticed that there were only ~18 different values 

```
3277,28
3515,63
3456,04
3155,96
3158,10
3513,24
3217,69
3453,70
3215,51
3275,06
3515,63
3456,04
3155,96
3158,10
3513,24
3217,69
3453,70
3215,51
```
[image]

Looking at the graph there are only about 5 values used in 100 samples.

smells like something systematic. (resolution of the math??)

RobTillaart, 2023-07-14:
The sample time is constant 1475 / 1476.
mmmm think think...

Can you show the code how you calculated the speed?
What is the unit of speed?  what does 3000 on the Y-axis represent?

JabberwockPL, 2023-07-14:
It is still your own example code: AS5600_angular_speed.ino, the only change I did was to get rid of the delay.

I still cannot figure out how to calculate how the periodicity relates to the rotation. If it matched, it might still indicate misalignment of the center of rotation/magnet/sensor, right?

RobTillaart, 2023-07-14:
> It is still your own example code: AS5600_angular_speed.ino, the only change I did was to get rid of the delay.

OK, 

#### Analysis

So **as5600.getAngularSpeed()**  returns between 2500 and 3500 degrees per second, (a)
when measuring every 1456 microseconds = ~700 measurements per second. (b)

a ==> at full speed the wheel is rotating roughly 10x per second (c)
 
b + c ==> 70 measurements per rotation ~ about every 5 degrees (d)

The sensor measures 4096 steps for 360 degrees, (e)

d + e ==> so 5 degrees is ~60 steps. ==> 2 digits precision here 

#### Proposal

Measure about 3x or 4x per rotation == 30x per second , should be fast enough to measure the speed of the wheel.
Then the wheel should make about 120 degrees between samples which increases to 4 digits.

Replace loop to something like this

```cpp
void loop()
{
  static uint32_t last = 0;

  if (millis() - last >= 25)  // 25 .. 33
  {
    last = millis();
    Serial.print(last);
    Serial.print("\t");
    Serial.println(as5600.getAngularSpeed());
  }
}
```

I have some other tasks this evening, so to be continued tomorrow (or maybe late tonight)

JabberwockPL, 2023-07-14:
With the changed code it still has the small spikes, but as I wrote, I can live with that - for my purposes such smal oscillations would be imperceptible.

RobTillaart, 2023-07-15:
> With the changed code it still has the small spikes, but as I wrote, I can live with that - for my purposes such smal oscillations would be imperceptible.

OK, still I would like to understand the source of those fluctuations but no high prio.

JabberwockPL, 2023-07-15:
Sure, I will keep on observing. Just let me know if you need something specific. I thought that maybe maintaining the speed at dffierent levels (4000, 2000, 1000) might give some clues concerning the math formulas?

RobTillaart, 2023-07-15:
Had some  thoughts about how the sensor might work inside. 
I share them here to document them so they are not lost, but are NO proven facts just thoughts.

Warning: thinking out loud section,

In essence every sensor is analogous, and this sensor has to decide which of 4096 "sub-sensors" get the strongest magnetic signal. Possibly this signal is not always equally strong and there might be an interference between the 4096 positions and the peaks of the magnetic signal. 

A second though is that the sensor might have some residual magnetic effect which differs per position. This could affect the absolute readings but I am far from sure how the signal would look like. This residual effect could even be "triggered" by the read process.

JabberwockPL, 2023-07-15:
I thought there are only only four subsensors? If so, a bias in one of them (or magnet misalignment) could have significant impact on the readouts.

RobTillaart, 2023-07-16:
> I thought there are only only four subsensors? If so, a bias in one of them (or magnet misalignment) could have significant impact on the readouts.

I should reread the datasheet if it mentions inner structure. I do not recall the details. But a bias in one of 4 or one in N , how would it create a roughly symmetrical sinus?

The frequency of this (fast) sinus should equal (assumption) or be an multiple (1 .. n) of the rotational frequency. IF it is caused by the rotation as every rotation same happens again. 

Mmm too early, need coffee :)

RobTillaart, 2023-07-16:
You're right, the datasheet shows 4 hall sensors 
It also describes that the displacement of the rotational axis should not be more than 0.25 mm which is pretty precise..

JabberwockPL, 2023-07-16:
I missed this one... Yes, you are right, that is quite precise, I doubt I can reach that accuracy in my setup.

RobTillaart, 2023-07-16:
> I doubt I can reach that accuracy in my setup.

maybe you need to use a small gearbox,

or use a different sensor. How about using an infrared beam and photocell?

Count the spokes of the wheel passing with interrupts.
Every spoke passing ==> 5 degrees have passed (72 spokes in a wheel IIRC)
From that one could calculate the distance and speed pretty straightforward. 
The interrupts occur every few milliseconds so with the micros() it is possible to see variation in the speed and e.g. difference between left and right leg.

Advantage is that alignment with the IR sensor is a no-brainer. 
Distance precision is 5 degrees which translates to ~2.5 cm (1 inch) == 1/72th of the circumference of the wheel.

idea?

RobTillaart, 2023-07-23:
@JabberwockPL 

If your bike is at rest, no movement  do you see the small fluctuations too?
Had some time to play with an AS5600L and it showed random noise if any, nothing close to a sine.

JabberwockPL, 2023-07-23:
Sometimes it had small noise after activation, but then it corrected itself to zero. I suspect the chip might autocorrect itself.

Also, I have decided to move to another solution, similar to what you describe. I still have the chip installed, but probably not for long, so I guess the issue might be closed.

But I am sure I will use the sensor in other ways...

RobTillaart, 2023-07-23:
OK, lets close the issue, if needed it can always be reopened, (or a new issue if a new problem arises)

Let me hear how your new solution works - maybe in this thread as it might be interesting for other people too.

closed 2023-07-23 by RobTillaart