RobTillaart/AS5600 #50: Reverse rotation gives incorrect (huge) numbers.

RobTillaart, 2023-12-10:
Rob, it is Mitch again, (Laptophead)

Having 2 sensors on one bus works. But there is still the issue of not getting good initial readings.

Despite the
as5600_R.resetPosition();
as5600_L.resetPosition();

when there are 2 sensors I get erroneous readings such as
REV R: -1
REV L: 0
DEG R= 377487200°
DEG L= 132°

Rolling the motors forward increases the angles as expected, but rolling them back past 0 deg. is
resulting in those huge numbers . Ideally it would go to 359 , 358 etc degrees ,

Would you know what to do about that?
Thanks
mitch

_Originally posted by @laptophead in https://github.com/RobTillaart/AS5600/issues/45#issuecomment-1849072350_

RobTillaart, 2023-12-10:
@laptophead

Created a new issue for it as was not related to #45 

Will look into this later this week

RobTillaart, 2023-12-10:
Please also

RobTillaart, 2023-12-12:
@laptophead

Any progress to report?

laptophead, 2023-12-14:
I am using the  convertRawAngleToDegrees(as5600_R.rawAngle());
It works better since I don't have to get more degrees than 360.

I am also having trouble with COUNTERCLOCKWISE,   it has no effect.
I made my own little algorithm,
 Deg_L = convertRawAngleToDegrees(as5600_L.rawAngle());
  Deg_L = 360 - Deg_L; //This will make the degrees increase as I go CW. (The AS5600 reads it backward)
  Deg_L = constrain (Deg_L, 0, 350);
Thanks for the nice library

RobTillaart, 2023-12-14:
Question:
What is the goal you want to achieve for the device that runs counter clock wise?
if it turns 90 degrees CCW, should it return 270 or 90?

The AS5600 is a position measuring device, so it works "like a compass".
There is a zero position and if you turn it CW it increases its position up to 4095.
If you turn it CCW it will go from zero to 4095 4094 4093 etc.

However if you set the direction from CW => CCW the direction of counting up changes. 
That does the sensor internally.

in short
- if a sensor is in the CW mode an you turn it 90 degrees CCW it will return 270 (~3000 raw)
- if a sensor is in the CCW mode an you turn it 90 degrees CCW it will return 90 (~1000 raw)

So maybe your problem is solved by setting both sensors in the CW mode.

RobTillaart, 2023-12-18:
@laptophead 
is your problem solved now?

RobTillaart, 2024-01-05:
@laptophead
I close this issue as I assume that no answer means it is solved. 
If not, feel free to reopen the issue

closed 2024-01-05 by RobTillaart

laptophead, 2024-01-05:
Rob
That issue is fixed but I have more trouble with the library,
If you want we could troubleshoot together and make it better, or I can learn what I am doing wrong.
1. setDirection(AS5600_CLOCK_WISE); does nothing, direction does not change when I go counter
2. as5600_R.setOffset(90); does nothing when I use as5600_R.getCumulativePosition()) but it works for raw. This is a useful feature.
3. Using the AS5600L I am getting weird readings, but consistent. 
0 to 360 are correct, but when I move it to 90 I get 126°.  When I move it to 180 it is correct. When I move it to 270 it shows 319, and at 360 is correct again.   Hardware problems? 

Let me know if you can help. Thanks for writing this.
Mitch

RobTillaart, 2024-01-06:
### Step one

First step is to read the datasheet until you understand the device you are working with.

### Step two

Second step is to connect the hardware and check it. 
- How is everything connected, 
- are the connections robust == soldered, tight breadboard
- are there special pins that configure the behavior? how are these set up?
- do you have pull up resistors where needed
- do you have a stable power supply.
these are things a library cannot handle / detect.

### Questions

If time permitted I would like to help anyone, however I have 200+ libraries to maintain with ditto issues, so time is a scarce resource. Best place to ask questions is the Arduino forum or stack exchange, there are dozens of people that can help you out.
Most time that will help you much faster if you provide all information they need, See below for the minimal info needed to analyze problems from remote.

#### Q 1 - setDirection
- please post a drawing of how you connected the sensor
- have you connected the hardware direction pin? how?
- please post a minimal version of the code that shows the effect.

#### Q 2 - getCumulativePosition()
Cumulative position is experimental, and your observation is correct.
To understand what it does, you need to read the code in the library and you will see it does not use the offset.
Then think what would happen if you would add offset for every angle measurement and add them up.

However you could change the last line into something like **return (_position + _offset);** Not tested for side effects.

#### Q 3 - weird readings
- please post a drawing of how you connected the sensor
- have you connected the hardware direction pin? how?
- please post a minimal version of the code that shows the effect.

As this issue (#50)  is solved, please make new issues - one per problem - if you want

laptophead, 2024-01-10:
OK
I see you're busy, I will work with what I got. 
For the direction pin I used
  as5600_R.begin(23);  //  set direction pin.
  as5600_R.setDirection(AS5600_CLOCK_WISE);  // default, just be explicit.
I soldered the wire from "DIR" to Pin 23 on my arduino DUE.  it is about 50cm long.  Not correct?

RobTillaart, 2024-01-11:
> as5600_R.begin(23); // set direction pin.
as5600_R.setDirection(AS5600_CLOCK_WISE); // default, just be explicit.
I soldered the wire from "DIR" to Pin 23 on my arduino DUE. it is about 50cm long. Not correct?

Should work, is the DUE a 3.3 volt device (don't know specs from my head). 
You might need a pull up resistor 4K7 or so on the wire to the DIR pin.