RobTillaart/AS5600 #51: setOffset() fails when called again

fahadshihab, 2023-12-31:
The `setOffset()` function does not work as expected when called more than once after the program starts. This makes it difficult to zero again when the program is running.

The current workaround is to use `setOffset()` in conjunction with `getOffset()` but the user has to implement the math for it.

RobTillaart, 2023-12-31:
Thanks for reporting the issue. 
I will look into it later this week

> The setOffset() function does not work as expected when called more than once after the program starts.

Can you elaborate, give examples what you expect?
e.g  setOffset(0) should it remove all offset or not change the current offset?

RobTillaart, 2023-12-31:
just a thought, do you need a function like

- **increaseOffset(float degrees)** increase or decrease the current offset with **degrees**

fahadshihab, 2023-12-31:
Suppose, at system start, the encoder reads 200 degrees.
Then when `setOffset(-200)` is called, the encoder now reads 0 degrees. 
Then the system is moved to, say, 30 degrees. The encoder now reads 30 degrees.
When `setOffset(-30)` is called, the encoder reads 200 degrees, instead of zero.

The problem is that the `setOffset()` function  sets the offset value directly instead of taking into account that the offset has been previously set.

fahadshihab, 2023-12-31:
Yes, something like `increaseOffset()` or `setReference(float)` could work.

`setReference(float angle)` could calculate the offset automatically and set the current encoder reading to `angle`. I feel it's a much more useful function when the readings must be reset multiple times in a program.

RobTillaart, 2023-12-31:
@fahadshihab 

I have created a develop branch which includes the **increaseOffset()** function.
Please verify if this meet your needs. If so I will merge it into version 0.5.1

RobTillaart, 2024-01-04:
@fahadshihab 
Did you verify the **increaseOffset()** function yet?

fahadshihab, 2024-01-05:
Yup. This works. Thanks!

commit 51346336 (2024-01-05): Fix #51, add increaseOffset() (#52)

- fix #51, add **increaseOffset(float degrees)**
- update keywords.txt
- update readme.md (several cleanups)
closed 2024-01-05 by RobTillaart