RobTillaart/AS5600 #57: Some examples marked as AS5600 use the AS5600L

Rob65, 2024-02-21:
While testing the AS5600 example, I could not get this to work.
Checking the I2C/Wire data with my scope showed that the slave address used was 0x40 - this is due to as5600 object being created from the AS5600L class instead of the AS5600.

I noticed this is also done in the following examples:
AS5600_angular_speed, AS5600_demo_ESP32, AS5600_demo_RP2040, AS5600_demo_status, AS5600_demo_STM32, AS5600_I2C_frequency and AS5600_resetCumulativeCounter examples.

RobTillaart, 2024-02-22:
Well noticed,
The library implements the AS5600L as derived class too. During tests I used an L version at some moment and they were left in the code.. I could have duplicated every example but that would lead to double maintenance (and longer build Ci times).

Will not fix this as it is easy adapted by the user.

Rob65, 2024-02-23:
Agree with not fixing but it might be nice to add a note on the main readme page to attent users to the fact that some examples use the 5600L. I tend to think I am fairly smart but it still to me a few minutes to notice this ;-)

RobTillaart, 2024-02-23:
Added comments on top of almost all examples:
```cpp
//  Examples may use AS5600 or AS5600L devices.
//  Check if your sensor matches the one used in the example.
//  Optionally adjust the code.
```

RobTillaart, 2024-02-23:
PR created
Build is running

closed 2024-02-23 by RobTillaart

rambros3d, 2025-03-13:
@Rob65 Thanks for creating this issue, I had no idea what was wrong. Silly me did not read the code comment

May I suggest something inline like this

```
// Uncomment the line according to your sensor
AS5600 as5600;   //  use default Wire
// AS5600L as5600;   //  use default Wire
```

RobTillaart, 2025-03-13:
@rambros3d 

Created a develop branch with updated examples.
Also added the line **while(!Serial)** as some boards take more time to start up serial communication,
Might be a point release after all

RobTillaart, 2025-03-15:
@rambros3d
0.6.5 released

commit cd55341f (2024-02-23): Fix #57, comments (#58)

Fix #57, update comments in examples