RobTillaart/AS5600 #59: Hardwired direction pin not used

LukeSkaff, 2024-02-25:
Thanks for making this great library.  I have a hardwired pullup on the direction pin on my AS5600 circuit board and I am setting the direction in the code to "AS5600_COUNTERCLOCK_WISE" to let the AS5600 code know.  There is no direction output from the Arduino to the AS5600 circuit board though.  There does not appear to be a way not to assign a direction output pin in the code, it looks like you have to use a direction pin.  Is there anyway around this?

RobTillaart, 2024-02-25:
Thanks for the issue
Might take a few days before I have time to dive into your question.

RobTillaart, 2024-02-26:
> There does not appear to be a way not to assign a direction output pin in the code, it looks like you have to use a direction pin. Is there anyway around this?

You should pass the direction pin in **begin()**

From the readme.md

----

#### DIR pin

From datasheet, page 30 - Direction (clockwise vs. counter-clockwise)

_**The AS5600 allows controlling the direction of the magnet 
rotation with the DIR pin. If DIR is connected to GND (DIR = 0)
a clockwise rotation viewed from the top will generate an 
increment of the calculated angle. If the DIR pin is connected
to VDD (DIR = 1) an increment of the calculated angle will 
happen with counter-clockwise rotation.**_

This AS5600 library offers a 3rd option for the DIR (direction) pin of the sensor:

1. Connect to **GND** ==> fixed clockwise(\*).  This is the default.
1. Connect to **VCC** ==> fixed counter-clockwise.
1. Connect to an IO pin of the processor == Hardware Direction Control by library.

In the 3rd configuration the library controls the direction of counting by initializing 
this pin in **begin(directionPin)**, followed by **setDirection(direction)**. 
For the parameter direction the library defines two constants named:

- **AS5600_CLOCK_WISE (0)**
- **AS5600_COUNTERCLOCK_WISE (1)**

(\*) if **begin()** is called without **directionPin** or with this 
parameter set to **255**, Software Direction Control is enabled.

See Software Direction Control below for more information.

----

LukeSkaff, 2024-02-26:
Thank you, sorry I missed that.

closed 2024-02-26 by LukeSkaff

RobTillaart, 2024-02-27:
No problem