RobTillaart/AS5600 #62: getCumulativePosition messed up by http.POST

ivukotic, 2024-03-17:
Hi,
not sure if this is a bug or a feature. I am reading getCumulativePosition from 8 sensors every 100 ms. 
After some time, I upload the data using http.POST (ESP Async Web Server). If sensor moved during the upload, getCumulativePosition skips (didn't notice the movement).
Best,
I

RobTillaart, 2024-03-17:
Thanks for the issue, 

The context:
```cpp
int32_t AS5600::getCumulativePosition()
{
  int16_t value = readReg2(AS5600_ANGLE) & 0x0FFF;     <<<<<<<<<<< (1)

  //  whole rotation CW?
  //  less than half a circle
  if ((_lastPosition > 2048) && ( value < (_lastPosition - 2048)))
  {
    _position = _position + 4096 - _lastPosition + value;
  }
  //  whole rotation CCW?
  //  less than half a circle
  else if ((value > 2048) && ( _lastPosition < (value - 2048)))
  {
    _position = _position - 4096 - _lastPosition + value;
  }
  else _position = _position - _lastPosition + value;   <<<<<<<<<<< (3)
  _lastPosition = value;

  return _position;
}

....

uint16_t AS5600::readReg2(uint8_t reg)
{
  _error = AS5600_OK;
  _wire->beginTransmission(_address);
  _wire->write(reg);
  if (_wire->endTransmission() != 0)
  {
    _error = AS5600_ERROR_I2C_READ_2;
    return 0;      <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< (2)
  }
  uint8_t n = _wire->requestFrom(_address, (uint8_t)2);
  if (n != 2)
  {
    _error = AS5600_ERROR_I2C_READ_3;
    return 0;  <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< (2)
  }
  uint16_t _data = _wire->read();
  _data <<= 8;
  _data += _wire->read();
  return _data;
}

```

**GetCumulativePosition()** does not check if  (1) the **readReg2()** call works or not. 
The ESP webserver may disrupt this I2C read (2) which returns 0 and does not see any movement.
If this happens a few times in a row, **GetCumulativePosition()** will read zero movement.
Then it will execute line (3)  with both _lastPosition and value zero ==> no movement.

No, this is not a feature, so it classifies as a bug ==> need to be fixed.

RobTillaart, 2024-03-17:
**GetCumulativePosition()** should check _error to catch the incorrect read.
The best it can do is 
- not update _lastPosition so it can be used when there is a correct read, assuming within half rotation.
- return _position.
- not clear the error state so you can check **lastError()** if **getCumulativePosition()** is updated or not, 
and optionally call it again.

in code this would add one line in the library.

```cpp
int32_t AS5600::getCumulativePosition()
{
  int16_t value = readReg2(AS5600_ANGLE) & 0x0FFF;
  if (_error != AS5600_OK) return _position;

  //  whole rotation CW?
  //  less than half a circle
  if ((_lastPosition > 2048) && ( value < (_lastPosition - 2048)))
  {
    _position = _position + 4096 - _lastPosition + value;
  }
  //  whole rotation CCW?
  //  less than half a circle
  else if ((value > 2048) && ( _lastPosition < (value - 2048)))
  {
    _position = _position - 4096 - _lastPosition + value;
  }
  else _position = _position - _lastPosition + value;   <<<<<<<<<<< (3)
  _lastPosition = value;

  return _position;
}
```

RobTillaart, 2024-03-17:
Can you verify if the proposed patch works for you?

ivukotic, 2024-03-17:
Hi,

thanks for the very quick reply. I tried your fix but it did not help. 
`_error != AS5600_OK`  is never true.

Best,
I

RobTillaart, 2024-03-17:
Mmm, 
Need to investigate if another scenario exists that can explain what happened. Will be difficult.

Can you add a check for lastError() after your calls to getCumulativePosition() to see if it fails and if so what the error code is.

ivukotic, 2024-03-17:
I added lastError call . It always return 0.
I

RobTillaart, 2024-03-18:
> I added lastError call . It always return 0.

So that excludes the scenario above. To be continued.

RobTillaart, 2024-03-20:
FYI,
not found a cause yet as it is unclear why there is not rotation seen.

Can you add a print statement in the function to see what values are read from the sensor?

```cpp
int32_t AS5600::getCumulativePosition()
{
  int16_t value = readReg2(AS5600_ANGLE) & 0x0FFF;
  Serial.println(value);
```

ivukotic, 2024-03-30:
Hi,
it is really hard to see from a printout as I have 8 sensors read every 100ms. 
but my application takes always covers the same angular range and I can plot min and max values of the range. Normally range is ~5900 counts. If I plot the values, I see that range increases:
[image]

RobTillaart, 2024-03-30:
Still can't explain the behaviour.

Could it be that the wifi signal distorts the sensor?

Have you tried to make a plot of raw reads?
That might help to see if it is a sensor problem .

ivukotic, 2024-03-30:
it is not wifi signal... When I turn off http post part and leave only part that sends data as events to the website (again via wifi), everything is fine. Here actual counts:
[image]
you may notice that normally I sleep 100ms after each read out. But then suddenly at 31:20.333 delay is 900 ms and count jumped outside normal range...

RobTillaart, 2024-03-31:
> it is not wifi signal... When I turn off http post part and leave only part that sends data as events to the website (again via wifi), everything is fine.

Too fast a conclusion imho,
these events probably have different data size, use other timing, which might be less interfering.

Q1: What data do you send over the HTTP POST?
Q2: How long does the POST take? 

So far I do see problems in the library (except for the read error not handled before).

Maybe it is time to post your question on the Arduino forum or so

RobTillaart, 2024-03-31:
Have you read #28 ?
might be same problem.

RobTillaart, 2024-03-31:
Created a develop branch + PR for the patch for **getCumulativePosition()** + some pending things.

ivukotic, 2024-04-01:
I will try suggestions from #28 and let you know what happens.

RobTillaart, 2024-04-11:
> I will try suggestions from #28 and let you know what happens.

Any new insights?

RobTillaart, 2024-04-16:
@ivukotic 
As there is no feedback, I assume the problem is solved so I close this issue.
Feel free to reopen it if needed.

closed 2024-04-16 by RobTillaart

ivukotic, 2024-04-20:
Hi Rob,
sorry for not commenting earlier. At the end I circumvented the issue by not posting from ESP32 but from the client side (client gets the same data via SSE).
best,
i

RobTillaart, 2024-04-21:
Good to hear you found a way to work around / solve the issue. Still I can not explain why / what happened that caused the issue in the first place.

Thanks