RobTillaart/AS5600 #64: rawAngle(), readAngle() only returns 0

k4pow, 2024-07-01:
I'm fairly new to ESP32 platform and was struggling to get an angle returned on your library. 

I've made sure my wiring was correct. I used another example code that returns raw values and it returns the expected 0 to 4096 and my magnet is approximately 6mm from the sensor.

When using your AS5600_demo_ESP32 example I can only get "0     0" as the output. I've even changed the SDA and SCL pins to match my wiring on my ESP32. 

I understand this may be a dumb question and my mistake is probably very silly, but any help is appreciated.

Thanks!
Ian

Below is the other example code that verified my wiring and setup was working

```
#include <Wire.h>
#include <Arduino.h>

#define AS5600_ADDRESS 0x36
#define AS5600_ANGLE_REG 0x0E

void setup() {
    Serial.begin(115200);
    Wire.begin(21, 22);  // Use your actual SDA and SCL pins

    Serial.println("Starting setup...");
    delay(1000);

    // Check if AS5600 is connected
    Wire.beginTransmission(AS5600_ADDRESS);
    if (Wire.endTransmission() == 0) {
        Serial.println("AS5600 detected.");
    } else {
        Serial.println("AS5600 not detected. Check wiring.");
    }
}

void loop() {
    // Read angle from AS5600
    Wire.beginTransmission(AS5600_ADDRESS);
    Wire.write(AS5600_ANGLE_REG);
    Wire.endTransmission();
    Wire.requestFrom(AS5600_ADDRESS, 2);

    if (Wire.available() == 2) {
        uint16_t angle = Wire.read() << 8 | Wire.read();
        Serial.print("Angle: ");
        Serial.println(angle);
    } else {
        Serial.println("Failed to read angle");
    }

    delay(1000);
}
```
And this code returns 
```
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Starting setup...
AS5600 detected.
Angle: 684
Angle: 668
Angle: 512
Angle: 618
Angle: 3712
Angle: 3666
Angle: 818
Angle: 2026
Angle: 2450
Angle: 1531
Angle: 1531
```

And here is your example code with the updated SDA and SCL Pins
```
//
//    FILE: AS5600_demo.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//     URL: https://github.com/RobTillaart/AS5600
//
//  Examples may use AS5600 or AS5600L devices.
//  Check if your sensor matches the one used in the example.
//  Optionally adjust the code.

#include <Arduino.h>
#include "AS5600.h"

AS5600L as5600;   //  use default Wire

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  Wire.begin(21, 22);

  as5600.begin(4);  //  set direction pin.
  as5600.setDirection(AS5600_CLOCK_WISE);  //  default, just be explicit.
  int b = as5600.isConnected();
  Serial.print("Connect: ");
  Serial.println(b);
  delay(1000);
}

void loop()
{
  if(as5600.detectMagnet()) Serial.print("TRUE\t");
  else Serial.print("FALSE\t");
  
  Serial.print(millis());
  Serial.print("\t");
  Serial.print(as5600.readAngle());
  Serial.print("\t");
  Serial.println(as5600.rawAngle() * AS5600_RAW_TO_DEGREES);
  delay(1000);
}

//  -- END OF FILE --
```

Your example returns
```
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
src/main.cpp
AS5600_LIB_VERSION: 0.6.1
Connect: 0
FALSE   1050    0       0.00
FALSE   2050    0       0.00
FALSE   3050    0       0.00
FALSE   4050    0       0.00
FALSE   5050    0       0.00
FALSE   6050    0       0.00
FALSE   7050    0       0.00
FALSE   8050    0       0.00
FALSE   9050    0       0.00
FALSE   10050   0       0.00
FALSE   11050   0       0.00
FALSE   12050   0       0.00
FALSE   13050   0       0.00
FALSE   14050   0       0.00
FALSE   15050   0       0.00
FALSE   16050   0       0.00
FALSE   17050   0       0.00
FALSE   18050   0       0.00
FALSE   19050   0       0.00
```

RobTillaart, 2024-07-01:
Thanks for your question.
Questions are never stupid, only not asking questions is (some philosopher said once).

The Connect == 0 means the device is not seen. So getting al zero's is as expected.

My guess is an address problem, try changing this line

```cpp
AS5600L as5600;   //  use default Wire
```
into 

```cpp
AS5600 as5600;   //   remove the L as that is another sensor, which has another default address.
```

RobTillaart, 2024-07-01:
See .h file

```cpp
//  default addresses
const uint8_t AS5600_DEFAULT_ADDRESS    = 0x36;
const uint8_t AS5600L_DEFAULT_ADDRESS   = 0x40;
```

k4pow, 2024-07-01:
That fixed it! Thanks!

RobTillaart, 2024-07-01:
You may close the issue

closed 2024-07-01 by k4pow