RobTillaart/AS5600 #65: getCumulativePosition not using direction

Chris-42, 2024-10-04:
getCumulativePosition does not use the direction set in initialization.
better replace
  int16_t value = readReg2(AS5600_ANGLE) & 0x0FFF;
by
   int16_t value = readAngle();
which is the same with taking direction into account.

RobTillaart, 2024-10-04:
Thanks for this issue,
Assuming you have tested this, did it gave better results? as I have no hardware nearby to verify.

I need to refresh my mind with the code details but it could be a good catch. 

----

(Note for myself)
When looking at **readAngle()** to check the direction code I noticed the **mod 4096** is done too often.
**rawAngle()** idem.

RobTillaart, 2024-10-04:
@Chris-42 

Can you verify this modification with your setup?
```cpp
uint16_t AS5600::rawAngle()
{
  int16_t value = readReg2(AS5600_RAW_ANGLE);
  if (_offset > 0) value += _offset;
  value &= 0x0FFF;

  if ((_directionPin == AS5600_SW_DIRECTION_PIN) &&
      (_direction == AS5600_COUNTERCLOCK_WISE))
  {
    value = (4096 - value);
  }
  return value;
}

uint16_t AS5600::readAngle()
{
  uint16_t value = readReg2(AS5600_ANGLE);
  if (_offset > 0) value += _offset;
  value &= 0x0FFF;

  if ((_directionPin == AS5600_SW_DIRECTION_PIN) &&
      (_direction == AS5600_COUNTERCLOCK_WISE))
  {
    value = 4096 - value;
  }
  return value;
}
```

Will make a develop branch to add your patch.

RobTillaart, 2024-10-04:
@Chris-42 

This is a related open issue.

Currently the function **getRevolutions()** returns -1 directly when the cumulative angle goes below zero. And -2 -3 etc 

Should it be more correct or intuitive to return zero until the first (negative) revolution has been made?

Chris-42, 2024-10-04:
think ist should be zero while in -4095 .. 4095

Chris-42, 2024-10-04:
and why not keep the  readReg2(AS5600_ANGLE) & 0x0FFF;
I found no register description stating the upper bits are always zero. Just to be on the safe side...

RobTillaart, 2024-10-04:
> and why not keep the readReg2(AS5600_ANGLE) & 0x0FFF; I found no register description stating the upper bits are always zero. Just to be on the safe side...

True, 
it only safes a few bytes and performance wise the gain is a few micros, where the I2C transaction is in the millis range.

Chris-42, 2024-10-04:
I copied the snipplet into my code, seems to work.

RobTillaart, 2024-10-04:
mmm, if the direction is "reverse" there should be a **& 0x0FFF** as 4096 - 0 = 4096 which should be 0.

Chris-42, 2024-10-04:
whats about introducing an update method which calculates speed and position? So the I2C register has only to be read once?
then one can have update called in loop and use siple getters for getCumulativePosition and getAngularSpeed.

RobTillaart, 2024-10-04:
**& 0x0FFF** Should be fixed in **setOffset()** too.

RobTillaart, 2024-10-04:
> whats about introducing an update method which calculates speed and position? So the I2C register has only to be read once? then one can have update called in loop and use siple getters for getCumulativePosition and getAngularSpeed.

The idea is interesting however not clear how it affects performance and footprint.
Please open a a new issue as I do not want to put too much new code into one version.

RobTillaart, 2024-10-04:
@Chris-42 

Created the develop branch for this issue (and some pending backlog). 
Build has started.

RobTillaart, 2024-10-04:
@Chris-42 

If you have time, can you verify that the develop branch fixes the use of direction for you?

RobTillaart, 2024-10-05:
Merged PR and published 0.6.2 release.
(only #67 pending, will pick that up asap)

commit d5f90ada (2024-10-05): Fix #65, getCumulativePosition() + some more (#68)

- fix #65, make **getCumulativePosition()** direction aware.
- refactor**readAngle()** and **rawAngle()**.
- fix negative values in **getRevolutions()**.
- fix #66, make I2C functions virtual to improve derived class.
- fix **setOffset()** to not set offset to 360.
- add unit test for offset -0.01 and 360.0 (both should become 0.0).
- add **AS5600_output_speedtest.ino**, thanks to Pollyscracker.
- update readme.md.
closed 2024-10-05 by RobTillaart