RobTillaart/AS5600 #66: make wire accessing functions virtual

Chris-42, 2024-10-04:
will be great to been able to overwrite the methods accessing the wire virtual.
So it will be possible in a derived class to easy access the sensor behind a mux.
  virtual bool     isConnected();
 ...
  virtual uint8_t  readReg(uint8_t reg);
  virtual uint16_t readReg2(uint8_t reg);
  virtual uint8_t  writeReg(uint8_t reg, uint8_t value);
  virtual uint8_t  writeReg2(uint8_t reg, uint16_t value);

I use it like this in my derived ENCODER class:
uint8_t ENCODER::readReg(uint8_t reg) {
  if(!_mux->setMuxChannel(_channel)) {
    _error = AS5600_ERROR_MUX;
    return 0;
  }
  xSemaphoreTake(i2cMutex, portMAX_DELAY);
  uint8_t ret = AS5600::readReg(reg);
  xSemaphoreGive(i2cMutex);
  _mux->freeMux();
  return ret;
}

RobTillaart, 2024-10-04:
Q: how does one knows the _channel?

Does the derived class provides a channel parameter in constructor or setter/getter function for the channel?

----

Not tested, however I think the setting of the I2C channel should also be under the mutex control.

Otherwise one can change channel in another thread before the actual read is done.

```cpp
I use it like this in my derived ENCODER class:
uint8_t ENCODER::readReg(uint8_t reg) {
  xSemaphoreTake(i2cMutex, portMAX_DELAY);
  if(!_mux->setMuxChannel(_channel)) {
    xSemaphoreGive(i2cMutex);
    _error = AS5600_ERROR_MUX;
    return 0;
  }

  uint8_t ret = AS5600::readReg(reg);
  xSemaphoreGive(i2cMutex);
  _mux->freeMux();
  return ret;
}
```

Chris-42, 2024-10-04:
it is only about to be allowed to overide...

RobTillaart, 2024-10-05:
@Chris-42 

I added the virtual keywords in the PR #68 
build is running.

commit 6b23449b (2024-10-05): fix #66
commit d5f90ada (2024-10-05): Fix #65, getCumulativePosition() + some more (#68)

- fix #65, make **getCumulativePosition()** direction aware.
- refactor**readAngle()** and **rawAngle()**.
- fix negative values in **getRevolutions()**.
- fix #66, make I2C functions virtual to improve derived class.
- fix **setOffset()** to not set offset to 360.
- add unit test for offset -0.01 and 360.0 (both should become 0.0).
- add **AS5600_output_speedtest.ino**, thanks to Pollyscracker.
- update readme.md.
closed 2024-10-05 by RobTillaart