RobTillaart/AS5600 #67: add update funktion for speed and cummulative position

Chris-42, 2024-10-04:
whats about introducing an update method which calculates speed and position? So the I2C register has only to be read once?
then one can have update called in loop and use siple getters for getCumulativePosition and getAngularSpeed.
maybe with a #define USE_UPDATE ...

`float AS5600::getAngularSpeed(uint8_t mode) {
  //  return radians, RPM or degrees.
  if (mode == AS5600_MODE_RADIANS)
  {
    return _angular_speed * AS5600_RAW_TO_RADIANS;
  }
  if (mode == AS5600_MODE_RPM)
  {
    return _angular_speed * AS5600_RAW_TO_RPM;
  }
  //  default return degrees
  return _angular_speed * AS5600_RAW_TO_DEGREES;
}

int32_t AS5600::getCumulativePosition() {
  return _position;
}

void AS5600::update() {
  uint32_t now  = micros();
  int16_t angle = readAngle();
  if (_error != AS5600_OK) return;
  // update cummulative position
  //  whole rotation CW?
  //  less than half a circle
  if ((_lastPosition > 2048) && ( angle < (_lastPosition - 2048))) {
    _position = _position + 4096 - _lastPosition + angle;
  }
  //  whole rotation CCW?
  //  less than half a circle
  else if ((angle > 2048) && ( _lastPosition < (angle - 2048))) {
    _position = _position - 4096 - _lastPosition + angle;
  }
  else {
    _position = _position - _lastPosition + angle;
  }
  _lastPosition = angle;

  uint32_t deltaT  = now - _lastMeasurement;
  int      deltaA  = angle - _lastAngle;

  //  assumption is that there is no more than 180° rotation
  //  between two consecutive measurements.
  //  => at least two measurements per rotation (preferred 4).
  if (deltaA >  2048) deltaA -= 4096;
  if (deltaA < -2048) deltaA += 4096;
  _angular_speed = (deltaA * 1e6) / deltaT;

  //  remember last time & angle
  _lastMeasurement = now;
  _lastAngle       = angle;

}
`

RobTillaart, 2024-10-04:
proposal forces the user to explicitly call **update()** first.

or should we check the timestamp of the last update call, and call update automagically?

Is definitely a behavior change. Need more thoughs.

Note: versioning => 0.7.0 ?

Chris-42, 2024-10-04:
you called the functions experimental, so why not changing in the next release. to keep the old behaviour you can use a #define
the timestamp may be also fine. no idea how much it influences the code size. I'm using esp32, so its not that big issue for me :)

RobTillaart, 2024-10-05:
> you called the functions experimental, so why ...

For me experimental means mainly not tested thoroughly. These functions are introduced 2.5 years ago which is an "eternity" in software terms.  As this is a very active library, downloaded/cloned ~8 times per day (platformIO + GitHub), there are many users, so breaking an interface needs a good reason (a bug is a good reason).

I have given the issue some thoughts last night and I think there is a way to increase the performance without breaking the interface. An extra parameter that indicates is the angle needs to be read again. Default true so it is backwards compatible,
setting it to false it uses the last value read to speed up the process. 

So I'm thinking in these lines. 
```cpp
float AS5600::getAngularSpeed(uint8_t mode, bool update = true) {
  if (update) _value = readAngle();
  ....
}

int32_t AS5600::getCumulativePosition(bool update = true) {
  if (update) _value = readAngle();
  ....
}
```

It would not break existing code while allowing the performance gain you had in mind. 
Code changes are quite straightforward which is less risky.

Chris-42, 2024-10-05:
Sounds good.

RobTillaart, 2024-10-09:
> only accessing the register once 

That is true, so I need to rethink it.

First I'm going to fix #69 as that is a bug, that has prio.

Chris-42, 2024-10-09:
I wrote it this:
`float AS5600::getAngularSpeed(uint8_t mode, bool update)
{
  if (update)
  {
    _lastReadAngle = readAngle();
    if (_error != AS5600_OK)
    {
      return nanf("");
    }
  }
  //  default behaviour
  uint32_t now     = micros();
  int      angle   = _lastReadAngle;
...
`
and this:
`int32_t AS5600::getCumulativePosition(bool update)
{
  if (update)
  {
    _lastReadAngle = readAngle();
    if (_error != AS5600_OK)
    {
      return _position;
    }
  }
  int16_t value = _lastReadAngle;
...
`

RobTillaart, 2024-10-09:
Fix for #69 merged and released, publish on platformIO, will be available within 24 hrs on Arduino library manager,

So now I can focus on this one

RobTillaart, 2024-10-09:
@Chris-42 

Working on the implementation, have written this test sketch to see the performance gain.

```cpp
//
//    FILE: AS5600_position_speed.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo compare performance with update flag
//     URL: https://github.com/RobTillaart/AS5600
//
//  Examples may use AS5600 or AS5600L devices.
//  Check if your sensor matches the one used in the example.
//  Optionally adjust the code.

#include "AS5600.h"

//  select right class.
//  AS5600 as5600;   //  use default Wire
AS5600L as5600;   //  use default Wire

uint32_t start, stop;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);

  Wire.begin();

  as5600.begin(4);  //  set direction pin.
  as5600.setDirection(AS5600_CLOCK_WISE);  //  default, just be explicit.

  Serial.println(as5600.getAddress());

  //  as5600.setAddress(0x40);  //  AS5600L only

  int b = as5600.isConnected();
  Serial.print("Connect: ");
  Serial.println(b);

  delay(1000);

  start = micros();
  as5600.getCumulativePosition();
  as5600.getAngularSpeed();
  stop = micros();
  Serial.print("update true: \t");
  Serial.println(stop - start);
  delay(100);

  start = micros();
  as5600.readAngle();
  as5600.getCumulativePosition(false);
  as5600.getAngularSpeed(AS5600_MODE_DEGREES, false);
  stop = micros();
  Serial.print("update false: \t");
  Serial.println(stop - start);
  Serial.println();
  delay(100);

  delay(2000);
}

void loop()
{
  static uint32_t lastTime = 0;

  if (millis() - lastTime >= 100)
  {
    lastTime = millis();
    as5600.readAngle();
    Serial.print(as5600.getCumulativePosition(false));
    Serial.print("\t");
    Serial.println(as5600.getAngularSpeed(AS5600_MODE_DEGREES, false));
  }
}

//  -- END OF FILE --
```

-----

Board: Arduino UNO, 
IDE:	1.8.19, 
I2C: 100kHz.

The output of the above sketch
```
Connect: 1
....\AS5600_position_speed.ino
AS5600_LIB_VERSION: 0.6.4
64
Connect: 1
update true: 	1200
update false: 	640

1521	0.00
1521	0.00
...
```

update true is the backwards compatible mode - takes 1200 µs
These calls:
```cpp
  as5600.getCumulativePosition();
  as5600.getAngularSpeed();
```

update false is your proposed optimized mode - takes 640 µs
These calls:
```cpp
  as5600.readAngle();
  as5600.getCumulativePosition(false);
  as5600.getAngularSpeed(AS5600_MODE_DEGREES, false);
```

Calling **readAngle()** updates the protected variable **_lastReadAngle**, which is used in the other two functions.

The performance gain is about 45% so substantial.
Note this number will differ on other boards and depends on I2C speed too.

The output of the **loop()** looks OK too.
```
1543	0.00
1543	0.00
1545	1.75
1546	0.88
1515	-27.43
1376	-121.74
1375	-0.88
1483	95.57
1767	248.71
1986	191.81
2239	223.88
2406	146.26
2632	199.98
2894	229.47
3004	96.33
2989	-13.27
2989	0.00
2988	-0.88
2988	0.00
2988	0.00
2988	0.00
```

Will create a develop branch and PR asap

RobTillaart, 2024-10-09:
@Chris-42 

Created a new develop branch and a PR, including the above test sketch.

RobTillaart, 2024-10-09:
@Chris-42 

Please verify if the develop branch meet your needs.

For me 
- there is a substantial performance gain, 
- the code is backwards compatible, 
- the interface did not get very complex to use that gain

To be continued tomorrow (for me)

RobTillaart, 2024-10-10:
@Chris-42 

Added a separate section in the readme.md to explain the optimization.  
Build is running,

RobTillaart, 2024-10-15:
@Chris-42 

Going to merge and release 0.6.4 now with the above improvement.
Thanks,

commit 0d01e335 (2024-10-15): Fix #67, performance getCumulativePosition and getAngularSpeed (#71)

- fix #67, improve performance **getCumulativePosition()** and **getAngularSpeed()**
  - kudos to Chris-42
- add **AS5600_position_speed.ino**
- update readme.md
closed 2024-10-15 by RobTillaart