RobTillaart/AS5600 #69: resetCummulativePosition not correct if direction set

Chris-42, 2024-10-08:
if one uses software direction the reset function for cummulative Position is not working.
in line 507 instead of using
`  _lastPosition = readReg2(AS5600_RAW_ANGLE) & 0x0FFF;`
it should be
`  _lastPosition = readAngle();`

RobTillaart, 2024-10-08:
Thanks for testing, appreciated.

Will dive into it tomorrow, looks more of the "same problem"

RobTillaart, 2024-10-09:
@Chris-42 
Develop branch is build,
Please give it a try

RobTillaart, 2024-10-09:
So the reset cumulative position now works.

GetAngularSpeed()
On update is false it returns the last calculated speed.

You can verify this by calling it once with true and second time with false.

Wrt your question, I want to find a form that is backwards compatible as the library is downloaded and forked a lot (>6x day).

To be continued.

Chris-42, 2024-10-09:
think this is fixed

RobTillaart, 2024-10-09:
Will merge the fix for this one into version 0.6.3 first.

commit bdd740f3 (2024-10-09): Fix #69
commit 9e0fc037 (2024-10-09): Fix #69, resetCumulativePosition() (#70)

- fix #69, **resetCumulativePosition()**
- minor edits
closed 2024-10-09 by RobTillaart