RobTillaart/AS5600 #76: Examples not working

JLBCS, 2025-03-25:
I open an issue on aarduino forum and on Simple FOC forum.
No answer so I open an issue her.
Ihave not been able to communicate correctly from a uno and/or a Nano ESP32 to 9 AS5600 normal size and 2 AS5600 mini.
None of the sample I tried work.
I mixed 2 examples and received the following:
`I2C Scanner
AS5600_LIB_VERSION: 0.6.5
Connect: 0
Scanning...
I2C device found at address 0x36  !
done

0	0
STATUS:	 0
CONFIG:	 0
  GAIN:	 0
MAGNET:	 0
DETECT:	 0
M HIGH:	 0
M  LOW:	 0

I2C Scanner
AS5600_LIB_VERSION: 0.6.5
`
the device is identified but does not return any valid data.
Here is the entire code:
`
`
Thank you for your help.

[AS5600_demo_status.txt](https://github.com/user-attachments/files/19454247/AS5600_demo_status.txt)

RobTillaart, 2025-03-25:
Thanks for opening the issue.

First question / remark is that the library support 2 devices AS5600 and AS5600L.
Given the address 0x36 you need to select the AS5600 in the sketches.

The code you attached uses the AS5600L class.
```
AS5600L as5600;   //  use default Wire
```

What board do you use?
How do you power the device and board?
How long are the I2C wires?
Do you have appropriate pull ups?

RobTillaart, 2025-03-25:
@JLBCS 
Do you have a link to the AS5600 normal size and AS5600 mini.

As I am not aware of different sized devices.

RobTillaart, 2025-03-25:
> I open an issue on aarduino forum

Can you provide a link, so I can read what is discussed there?

JLBCS, 2025-03-25:
Hi,
as you suggested I changed 2 lines of the example:
`//AS5600L as5600;   //  use default Wire
AS5600 as5600;`
Now it work fine.
Need to warn and document more clearly.
Thnak you for your help.

RobTillaart, 2025-03-25:
> Need to warn and document more clearly.

From the file you send me in the header

```
//  Examples may use AS5600 or AS5600L devices.
//  Check if your sensor matches the one used in the example.
//  Optionally adjust the code.
```

> Thnak you for your help.

You are welcome

RobTillaart, 2025-03-25:
If the problem is solved, you may close the issue.

RobTillaart, 2025-03-25:
@JLBCS

Checked the examples and 21 of the 25 has the note of the two device types in the header.

closed 2025-03-25 by JLBCS

JLBCS, 2025-03-31:
You are right, there are comments.
However, I suggest that you add a config checker where there is neither AS5600 nor AS5600L indicated so the exaample will not compile and that will catch the attention of the user.
Secondly, in case of wrong device (isconnect()!= 1), issue a message about the device that could be wrong even thought the address is correct.
Here is the code for AS5600_config_checker

[AS5600_config_checker.txt](https://github.com/user-attachments/files/19537545/AS5600_config_checker.txt)

RobTillaart, 2025-03-31:
I understand your idea, however it is impossible to make such checker 100% fool-proof.
Furthermore the extra code would make the examples more complex or if build into the library would increase the footprint.
And to be honest, this library now exist for 3 years and most people don't have problems with fixing the examples.

OK, that said.

Scanning for an address can also detect other I2C devices e.g. LCD or whatever, so not 100% foolproof.
The address of the AS5600 is fixed to 0x36 and the AS5600L is default 0x40 so those two addresses can be tested.
If those do not match one may assume that the device is an AS5600L with a different address set.

Something like:

```cpp
//
//    FILE: AS5600_detect_type.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo
//     URL: https://github.com/RobTillaart/AS5600
//
//  Examples may use AS5600 or AS5600L devices.

#include <Wire.h>
#include "AS5600.h"

//  if address of the AS5600L has been changed it can be any value including 0x36 !!!
//  AS5600 == 0, AS5600L == 1;  undefined == 2
int as5600Type = 2;

void setup()
{
  while (!Serial);
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("AS5600_LIB_VERSION: ");
  Serial.println(AS5600_LIB_VERSION);
  Serial.println();

  Wire.begin();

  Serial.println("check address AS5600 0x36: ");
  Wire.beginTransmission(0x36);
  int error = Wire.endTransmission();
  if (error != 0)
  {
    Serial.print("not found.");
  }
  else
  {
    Serial.println("found.");
    as5600Type = 0;
  }

  Serial.println("check address AS5600L 0x40: ");
  Wire.beginTransmission(0x40);
  error = Wire.endTransmission();
  if (error != 0)
  {
    Serial.print("not found.");
  }
  else
  {
    Serial.println("found.");
    as5600Type = 1;
  }

  if (as5600Type == 2)
  {
    Serial.println("Could not determine type, so set it to AS5600L");
    as5600Type = 1;
  }
}

void loop()
{
  if (as5600Type == 0)
  {
    AS5600 device;
    device.begin();

    Serial.print(device.readAngle());
    Serial.print("\t");
    Serial.println(device.rawAngle());
    Serial.print("STATUS:\t ");
    Serial.println(device.readStatus(), HEX);
    Serial.print("CONFIG:\t ");
    Serial.println(device.getConfigure(), HEX);
    Serial.print("  GAIN:\t ");
    Serial.println(device.readAGC(), HEX);
    Serial.print("MAGNET:\t ");
    Serial.println(device.readMagnitude(), HEX);
    Serial.print("DETECT:\t ");
    Serial.println(device.detectMagnet(), HEX);
    Serial.print("M HIGH:\t ");
    Serial.println(device.magnetTooStrong(), HEX);
    Serial.print("M  LOW:\t ");
    Serial.println(device.magnetTooWeak(), HEX);
    Serial.println();
  }
  else  //  default AS5600L
  {
    AS5600L device;
    device.begin();

    Serial.print(device.readAngle());
    Serial.print("\t");
    Serial.println(device.rawAngle());
    Serial.print("STATUS:\t ");
    Serial.println(device.readStatus(), HEX);
    Serial.print("CONFIG:\t ");
    Serial.println(device.getConfigure(), HEX);
    Serial.print("  GAIN:\t ");
    Serial.println(device.readAGC(), HEX);
    Serial.print("MAGNET:\t ");
    Serial.println(device.readMagnitude(), HEX);
    Serial.print("DETECT:\t ");
    Serial.println(device.detectMagnet(), HEX);
    Serial.print("M HIGH:\t ");
    Serial.println(device.magnetTooStrong(), HEX);
    Serial.print("M  LOW:\t ");
    Serial.println(device.magnetTooWeak(), HEX);
    Serial.println();
  }
  delay(3000);
}
```