RobTillaart/AS5600 #78: How works getDirection

Guenni75, 2025-05-09:
Hi.

I try to figure out how getDirection works.

```c++
#include <AS5600.h>

#define SDA_PIN1 18  // Beispiel: GPIO 21 für SDA
#define SCL_PIN1 19  // Beispiel: GPIO 22 für SCL
#define SDA_PIN2 16
#define SCL_PIN2 17

//#define DIR_PIN1 22
//#define DIR_PIN2 21

TwoWire I2C_1 = TwoWire(0);  // First I2C bus
TwoWire I2C_2 = TwoWire(1);  // Second I2C bus

AS5600 as5600_1(&I2C_1);
AS5600 as5600_2(&I2C_2);

void setup() {

  Serial.begin(115200);
  I2C_1.begin(SDA_PIN1, SCL_PIN1);  // SDA, SCL for first sensor
  I2C_2.begin(SDA_PIN2, SCL_PIN2);  // SDA, SCL for second sensor
  as5600_1.begin();
  as5600_2.begin();
  as5600_1.setDirection(AS5600_CLOCK_WISE);  //  default, just be explicit.
  as5600_2.setDirection(AS5600_CLOCK_WISE);  //  default, just be explicit.
  as5600_1.resetPosition();
  as5600_2.resetPosition();
}

void loop() {
  Serial.print("Sensor 1: ");
  Serial.print(as5600_1.getCumulativePosition());
  Serial.print(" - ");
  Serial.print(as5600_1.getDirection());
  Serial.print(" | Sensor 2: ");
  Serial.print(as5600_2.getCumulativePosition());
  Serial.print(" - ");
  Serial.print(as5600_2.getDirection());

  Serial.println();
  delay (100);
}
``` 

I only get zero's.

RobTillaart, 2025-05-09:
@Guenni75 
Thanks for the quiestion,

Can you check if the devices are seen on their I2C bus?

```cpp
 Serial.println( as5600_1.begin() );
 Serial.println( as5600_2.begin() );
```

wrt **getDirection()**

it just returns what is configured by **setDirection()**
```cpp
const uint8_t AS5600_CLOCK_WISE         = 0;  //  LOW
const uint8_t AS5600_COUNTERCLOCK_WISE  = 1;  //  HIGH
```

as you have set **AS5600_CLOCK_WISE** that internal parameter will be zero.

The meaning of **setDirection()** is to define which direction is counting up / down.

RobTillaart, 2025-05-09:
As described in the readme.md, use
 GetAngularSpeed(). 

It will give positive or negative values depending on the direction.

Guenni75, 2025-05-09:
Thanks.
I will try it tomorrow.

Guenni75, 2025-05-10:
Hi.
That works. 
Thank.

closed 2025-05-10 by Guenni75